OrcaSlicer/OrcaSlicer #3589: Separate Bed Options from Filament

shortbus, 2024-01-09:
**Is your feature request related to a problem? Please describe.**
There should be a whole separate section for Bed settings. Each Plate should have all the settings associated with that plate all together. When I chose the bed, all the settings should be assumed to be those of the bed/plate chosen.

There is no reason one should go into filaments and set the bed temp. Why are all the beds listed there? There should be a Plate selection and a settings page for that Plate.  One can choose the first layer temps etc. from that settings page.

**Which printers will be beneficial to this feature**
All Printers

**Describe the solution you'd like**
A separate set of settings like filaments have but for all things Bed/Plate related.

shortbus, 2024-01-09:
Honestly, going into every filament to change the bed temp settings is insane.

igiannakas, 2024-01-09:
Dont agree because the bed temperatures vary by filament type and by bed type. For example, a cool plate needs significantly lower temps than a PEI plate and the PEI plate different temperatures to the engineering plate. And both temperatures are different depending on what material you print - PLA needs lower temps compared to ABS.

shortbus, 2024-01-09:
It still works to have an Engineering PLA settings and Engineering PETG settings, etc. Having to go through all filament settings to change the plate temp is the wrong. If you have a PEI plate that needs more temp, then a different PEI plate then one has to go through all the filament settings to change the plate temp... this is the wrong place for it. So, by your claim -- if I have 5 PLA filaments with 5 different bed temp settings in their configuration, then the temperature of the bed changes every time the filament changes? The bed temp doesn't need changing at layer 500. A plate temp profile should have the bed temp for the entire print. It should not change. Setting up different profiles for different plates can be a thing.

shortbus, 2024-01-09:
One small thing, I would not mind seeing an Initial Bed temp, then a different bed temp that gets set at X layer. (for something like the initial bed temp at 65C, then at layer 4 set it to 60C) But the bed temp should not change per filament.

closed 2024-04-17 by github-actions[bot]