OrcaSlicer/OrcaSlicer #3590: Can't install

MadsSte, 2024-01-09:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Additional system information

Completely fresh Win11 installation

### Printer

Bambu x1c

### How to reproduce

I downloaded the installation file and tried to install it. 
Afterwards the portable file. Both gave me this error.

The version 1.8.1 installs fine. and the newest Bambu Studio has also been installed without a problem.

### Actual results

[image]

After getting this error, I click yes, and then the windows store can't find anything

### Expected results

I expected the program to be installed :*)

### Project file & Debug log uploads

don't have any

### Checklist of files to include

- [ ] Log file
- [ ] Project file

1ambo, 2024-01-13:
v1.9.0 Windows Installer also failing to install on Windows 11 Pro - Version	10.0.22631 Build 22631

Only difference for me was there was no error to screenshot. Absolutely nothing appears to happen.

Have v1.6.6 currently installed.

lech91, 2024-01-21:
how do i solve this installer not working in windows 11?

berri3D, 2024-02-04:
I experienced the same problem as @1ambo  when I went to install 1.9.0 on Windows 10 today Version 22H2 OS build 19045.3930.

Nothing happened when I launched the installer. I tried running it as an Administrator. Again, nothing happened.

After rebooting my machine a few times and downloading the installer again, I was able to get it to install by enabling Windows 8 compatibility mode.

Right-click installer -> Troubleshoot Compatibility -> Try recommended settings

BearScare13, 2024-03-10:
Hello, I am also having this problem. Windows 11 legion 5 pro laptop. Ive tried 1.9.0 and 1.9.1 also tried compatibility mode but all the same result : nothing happens, nothing opens, cpu stays at 0%

A crash file forms but all it says is "appcrash" for the event

prusa, cura, bambu, and lychee are all installed and work fine.

closed 2024-06-17 by github-actions[bot]