OrcaSlicer/OrcaSlicer #3594: High GPU usage when color scheme is set to "Filament" or "Speed"

Snelso91, 2024-01-09:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10

### Additional system information

CPU: AMD Ryzen 7 3700X
RAM: 32GB
GPU: Nvidia RTX 3060

### Printer

Bambu Lab A1

### How to reproduce

1. Slice a print
2. Change color scheme to "Filament" or "Speed"

### Actual results

The GPU usage should be relatively low when viewing the sliced print with the default color scheme "Line Type" or any of the other schemes apart from filament or speed.
However, when filament or speed is selected, the GPU usage increases from about 1% max to around 48% max when not interacting with OrcaSlicer.
This occurs even when the window is not focussed and is not moused over/being interacted with, and also occurs when the window is minimised.
The high usage does not occur if a tab other than "Preview" is selected, even if filament or speed remains selected on the preview tab.

### Expected results

The GPU usage should remain at the same level as when using all the other color schemes.
If this is not possible then the filament or speed color schemes should at least not have high GPU usage when OrcaSlicer is minimised or the window is not focussed.

### Project file & Debug log uploads

[cube gpu usage bug.zip](https://github.com/SoftFever/OrcaSlicer/files/13880884/cube.gpu.usage.bug.zip)
[debug_Tue_Jan_09_22_43_35_26484.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13880885/debug_Tue_Jan_09_22_43_35_26484.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

killerline10168, 2024-01-12:
I have been experiencing the same thing, although since I started using Orca (1.6.6). No issues whatsoever on all preview pages, except when viewing **Speed** and especially **Flow**, then it is a stuttering mess.

I have attached a video of how it looks.

https://github.com/SoftFever/OrcaSlicer/assets/1812366/d8a3301a-7e34-42dc-a489-68e87f281bb3

CPU: Ryzen 5 5600X AIO Cooled
GPU: AMD Radeon RX 6950 XT (driver version: 23.12.1 (latest))
RAM 32 GB DDR4 3800 Mhz CL16
Storage: NVMe PCIe gen 3 SSD

abortz, 2024-01-12:
Can confirm the issue when "Speed" or "Filament" is selected. Doesn't seem to happen for me with "Flow". I see both increased CPU and GPU utilization even when Orca is idle and not focused.

Windows 10
Orca 1.9.0 release
CPU: i5-12400F
GPU: RTX 3070
RAM: 16GB

nirurin, 2024-01-23:
I actually thought I had a crypto miner or some kind of other malware on my system when my 3D modelling software slowed down to a crawl and my 3090 was pegged to 100%. 

What on Earth is orcaslicer doing that utilises 100% of a 3090??

Felix14-v2, 2024-01-27:
Can confirm, I noticed that extreme GPU usage with both intel UHD 630 and GTX 1050. This is critical issue, because the entire system dies when this panel is opened.
Oh, correction to the previous reports: GPU usage increases only when the color scheme panel opened, collapsing it "fixes" the issue.
Demo:
https://youtu.be/iL4akoPdn8I

[image]
[image]

kylek29, 2024-02-02:
Can confirm, same on Windows 10 w/ Nvidia GPU.

[image]

[image]

vgdh, 2024-03-19:
confirm
win10
nvidia rtx460

BeauSlim, 2024-04-11:
Same.  Macbook Air M1 on Ventura.  2.0.0 release.  My laptop's (3rd party) temperature alarm was beeping at me even though I hadn't touched the machine for a few minutes.  GPU pegged to 100% even when Orcaslicer is off-screen.

NickUfer, 2024-07-11:
Push, still an issue

Felix14-v2, 2024-07-11:
Can confirm in 2.1.1

Snelso91, 2024-07-13:
I can confirm this is still an issue on the v2.1.1 official release.

This is particularly noticeable when slicing any multi colour or multi material prints, as the slicer always defaults to the "filament" colour scheme in this case, and so ends up causing high utilisation on the GPU and therefore causing unnecessary power usage as well as heat and noise.

PR #6223 (2024-07-25): Fixing high GPU usage when color scheme is set to "Filament" or "Speed"
First time contributing, I hope I'm doing this correctly.

# Description
Removed `imgui.set_requires_extra_frame()`, this line causes Slic3r::GUI::GLCanvas3D::render() to be called repeatedly when 'Filament' or 'Speed' is selected in the preview panel (even when OrcaSlicer is minimized)

This should fix #3594 

I see that this line was added in #3363. I don't fully understand the intention behind it, but removing this line does not seem to affect the legend window fix

# Screenshots/Recordings/Graphs
Now in "Filament" or " [... cut]
closed 2024-07-25 by SoftFever