OrcaSlicer/OrcaSlicer #3596: Allow rotation of Wipe Tower

eschreiner, 2024-01-10:
**Is your feature request related to a problem? Please describe.**
I have one large object to print where the prime tower only fits into a corner if rotated by 45°

**Which printers will be beneficial to this feature**
All with MMU

**Describe the solution you'd like**
The setting seems to be half way implemented as the search already knows about it, but the related controls are invisible (or I am too stupid). I am using a Bambu Lab P1S, but there seems to be no good reason as to why this setting should not be available here.

**Describe alternatives you've considered**
Hacking the 3MF file did not work :(

**Additional context**
see screenshot taken from Orca Slicer 1.9.0

[image]

streetgt, 2024-02-06:
+1 for this, would be great to have more options to control the Prime Tower location.

apadua, 2024-02-21:
Agreed. Important feature.

CristianGTR97, 2024-03-29:
THIS WOULD BE VERY USEFUL

Gozania, 2024-06-03:
+1 for this... Bonus points if you just give us full control of the prime tower... (Placement wherever we want & positioned however we want.

exsodus2, 2024-06-29:
I just ran into a print where this would have been very helpful. I had to cover calibration lines because I couldn't orient the tower. +1 for this feature

Magic815, 2024-08-10:
Just ran into a print bed where being able to rotate the wipe tower would have been the difference between fitting all parts on one plate, versus having to spread them across two plate. +1 for this feature!

Gozania, 2024-08-10:
Its truly asinine that full control of the prime tower hasn't been implemented yet....   Its obviously a feature that many of us want.

wormiedk, 2025-01-16:
It seems to be in orca slicer 2.2.0 but for me it has no effect

[image]

Jazun713, 2025-01-16:
doesn't seem to be working for 2.3.0-dev either.

hugues974974, 2025-04-12:
Yes this problem is still there.... Prime tower can be turned in Prusa Slicer but doesn't work in ORCA.
It is very usefuk to give that prime tower an angle for large prints. 
Awaiting for a correction in a last version ???
Best regards,
Hugues

Magic815, 2025-07-12:
Would still love to see this feature working

Kathode-Ray, 2025-11-14:
Wipe tower rotation angle parameter doesn't do anything. I have set it to 45 degress, but it stays stuck at the shown position.

[image]