OrcaSlicer/OrcaSlicer #3598: Slow down for curled perimeters is overriding bridge speeds with overhang speed (75%,100%)

sleewok, 2024-01-10:
### OrcaSlicer Version

2.0.0-dev

### OS version

Windows 11

### Printer

Ender 3 Pro w/Klipper

### How to reproduce

Load a model with a bridge. 

Enable the following: 

- Slow down for overhangs
- Slow down for curled perimeters

Slice the model.

### Actual results

The bridge speed is ignored and the overhang speed is used. 
[image]

Disabling "slow down for curled perimeters" allows bridges to run at the defined speed.
[image]

### Expected results

The bridge speed should be used since it is a bridge and not an overhang.

### Project file & Debug log uploads

[bridge-test.zip](https://github.com/SoftFever/OrcaSlicer/files/13890410/bridge-test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

sleewok, 2024-01-10:
#2056

igiannakas, 2024-01-10:
[image]

Opened your project - cannot replicate. The perimeter is printed with the slowdown speed and the bridge is printed with the bridge speed with slowdown for curled overhangs enabled. The perimeter of the bridge is not classed as a bridge perimeter but rather an overhang perimeter and will always be printed with the slowdown for overhangs value when the curled algorithm is enabled. The reason for this is because when you're printing shapes that benefit from this feature (i.e. with overhangs) you dont want the printer accelerating when the overhang increases and the perimeter prints in mid air.

Like this for example:

[image]

vs. 

[image]

closed 2024-04-17 by github-actions[bot]