OrcaSlicer/OrcaSlicer #3603: Retractions in vase mode when Travel distance threshold set to 0

Richard-Wals, 2024-01-11:
### OrcaSlicer Version

1.9.0

### OS version

OS: Windows 10

### Printer

Snapmaker J1S

### How to reproduce

1. Add a primitive cylinder
2. Enable spiralize vase
3. Set Travel distance threshold to 0 in either machine settings or filament settings
4. Slice

### Actual results

[image]

### Expected results

[image]
There are no travel moves in vase mode (at least after the initial layers) so there shouldn't be any retractions. 

### Project file & Debug log uploads

[Cylinder.zip](https://github.com/SoftFever/OrcaSlicer/files/13895937/Cylinder.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-01-11:
I've attempted to resolve this here: https://github.com/SoftFever/OrcaSlicer/actions/runs/7489048846

Please download the latest build from the link above and report back if this is resolved.

Richard-Wals, 2024-01-11:
Yeah this fixes the problem, thank you. I just had a little poke around and this also occurs in Prusaslicer, but not Superslicer.

PR #3614 (2024-01-15): [Bug fix] Fixes retraction on layer change being performed when retract travel threshold is set to 0
Fixes #3603 

Before:
[image]

After
[image]

Before
[image]

After:
[image]
closed 2024-01-15 by SoftFever