OrcaSlicer/OrcaSlicer #3605: Reset of G-code thumbnails in Printer:Basic information does not work

paul95831, 2024-01-11:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10

### Printer

Elegoo Neptune 4

### How to reproduce

Printer settings --> G-code thumbnails

Changing the value of this field, then pressing the now-displayed reset icon has no effect.  Does not reset the field contents.

### Actual results

Attempting to reset the "G-code thumbnails" field to it's original by pressing the reset icon has no effect.

### Expected results

Pressing the reset icon should reset the field to it's original value, then remove the reset icon.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

danshoop, 2024-02-16:
There appears to be a regression from the beta releases (when tis was working) and the output is in the incorrect format from what's selected.

danshoop, 2024-02-16:
This should be closed as it appears to be a duplicate of https://github.com/SoftFever/OrcaSlicer/issues/3320

closed 2024-05-23 by github-actions[bot]