OrcaSlicer/OrcaSlicer #3606: Device Tab - Connection to printer (Bambu X1C) is flapping in 1.9.0

WebbuffUSA, 2024-01-11:
### Discussed in https://github.com/SoftFever/OrcaSlicer/discussions/3604

<div type='discussions-op-text'>

<sup>Originally posted by **WebbuffUSA** January 10, 2024</sup>
After upgrading to 1.9.0 I get the following error when under the Device tab of Orca Slicer 1.9.0:"Failed to connect to printer"

This connection to the device is "flapping" from connected to not connected (with the error) so it is not a constant state.

I am able to slice and send prints to the printer and the printer (X1C) prints the file fine.

I have no issues with the Handy app for this printer, nor is my wife's laptop having the same issue.

When I re-install 1.8.1, the issue does not occur. Here are the steps I have taken to attempt to resolve this issue.
- Rebooted the printer multiple times.
- Uninstalled and re-installed 1.9.0.
- Deleted the plugins folder in the appdata folder to force the network plug-in to reinstall.
- Removed (forgot) wifi connection to PC and re-added.
- Logging out and back into the printer from Orca Slicer
- Uninstalled 1.9.0 completely and deleted any remaining folders and re-installed.
- Installed 1.8.1 and re-tested (works fine).
- Checked for other reports of similar issues on Reddit/Google.
- Ensured firewall had the proper exception for this application (it did).
- Started application and waited 60 seconds before moving to the Device tab.

Not sure what else to try or if this is just a bug and I need to wait for next patch.</div>

mattsfalco, 2024-01-11:
Same issue here running on M1 Mac. Device tab is just a white box where nothing loads.

WebbuffUSA, 2024-01-11:
In my case, the Device (X1C) is listed, it just goes back and forth (ever 2-3 seconds) from connected to not connected (with the described error). Had someone in Reddit mention a similar issue as yours and they mentioned waiting 30 seconds after opening Orca Slicer before navigating to the device page.

https://www.reddit.com/r/OrcaSlicer/s/HEE5QrbLzd

JoseMHRuiz, 2024-01-20:
Same problem here, back to 1.8.1, I already tried the same things as @WebbuffUSA. Not working

sudo-kraken, 2024-02-20:
same issue here with version 1.9.1

WebbuffUSA, 2024-02-27:
Weirdly enough, it stopped occurring after not using Orca Slicer for a couple weeks. I then update to 1.9.1 and was fine for a day, but now the issue has returned. I re-tried disconnecting and reconnecting the printer in Orca and also deleting the Plugins folder and re-installing the Network Plug-in. No change.

closed 2024-06-05 by github-actions[bot]