OrcaSlicer/OrcaSlicer #3615: Zooming into side views causes the view to disappear

liftbag, 2024-01-11:
### OrcaSlicer Version

1.9.0 official

### OS version

macOS Sonoma 14.3 Beta 3

### Additional system information

CPU: Apple M2 Pro
Memory: 32 GB
Display card: Apple M2 Pru GPU 19 core (Metal 3 supported)

### Printer

Not relevant, slicer issue.

### How to reproduce

The problem occurs both in the prepare and in the preview tabs.

1. Open the slicer
2. Click on New project
3. Go to any side view, either via the View menu or via the modifier keys (shortcut 3, 4, 5, 6)
4. zoom with the mouse wheel

### Actual results

The bed disappears from sight and never reappears.
This also happens with a bed populated with objects.
This does not happen zooming from the top and bottom views.
The problem happens in both perspective and orthogonal views.

### Expected results

Being able to freely zoom without losing the view of the printing surface.
In my case the behavior is systematic, and forces me to close and reopen the slicer.
I had previously notified a developer on Discord, as I believed it was a specific issue with his builds. But I found out that the latest release also has this bug.
I am attaching an empty project file, so you can evaluate if there was any setting that triggers the problem.

### Project file & Debug log uploads

[Empty.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13904095/Empty.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2024-01-11:
Yes finally!
Thank you for reporting this.
This is a real show-stopper since there is no way to recover from this state. You basically loose all the changes as you have to restart the program.

@SoftFever @Noisyfox @igiannakas @Ocraftyone

thanks in advance

discip, 2024-01-11:
Yes finally!
Thank you for reporting this.

This is a real showstopper as there is no way to recover from this state. Basically, all changes are lost as you have to restart the program.

@SoftFever @Noisyfox @igiannakas @Ocraftyone

thanks in advance

igiannakas, 2024-01-12:
Can't replicate it on 14.2.1 M1 Pro. Any tips?

https://github.com/SoftFever/OrcaSlicer/assets/59056762/8d48823c-7c0e-4d02-a143-6803437465bc

liftbag, 2024-01-12:
> Can't replicate it on 14.2.1 M1 Pro. Any tips? Same with the above project file.
> 

Where do I find the setting to switch from dark to light mode? Although very unlikely, I want to replicate your usage settings, although I don't want to downgrade to 14.2. However, I know that @Noisyfox has the same anomalous behavior.

liftbag, 2024-01-12:
In this video, you see that something happens at a certain point. The bed slides very far and it is no longer possible to take a close up at the maximum zoom level. When the view disappears it is because I activated the command+5 keyboard shortcut.

https://github.com/SoftFever/OrcaSlicer/assets/14921654/0d4def2c-4f92-4f6d-8333-a7d8059ecb94

liftbag, 2024-01-12:
To further integrate, OrcaSlicer 1.8.1 works fine.

glebsexy, 2024-01-13:
I can confirm, also have this issue. macOS 12.6.7, Intel MBP.

liftbag, 2024-01-14:
> I can confirm, also have this issue. macOS 12.6.7, Intel MBP.

I would say that the macOS version can be excluded.

glebsexy, 2024-01-14:
I just tested: if you zoom in by pinching the trackpad it  works but if you zoom by scrolling it breaks.

liftbag, 2024-01-14:
It's starting to be a limiting issue from my point of view. On the bright side, it's teaching me to save my work often :)

discip, 2024-01-14:
@igiannakas
> Can't replicate it on 14.2.1 M1 Pro. Any tips? Same with the above project file.

1. Press **`Ctrl + (one of 3 - 6)`**.
2. Use your mouse wheel to zoom.

It is still present in the latest builds both for Win & Linux (Manjaro).

liftbag, 2024-01-14:
I set the mouse wheel to scroll freely (not in steps) to better evaluate when the glitch occurs. It seems that the viewport collapses when the bed is moved away to almost the maximum distance.

https://github.com/SoftFever/OrcaSlicer/assets/14921654/7659d0d2-ac9a-4cda-a18c-e3cf3bb456c3

libfitz, 2024-01-18:
I have the same issue happening on macOS 14.2.1, Orca 1.9.0, although for me it happens regardless of whether I zoom using a mouse or trackpad.

Noisyfox, 2024-01-18:
I think this can be reproduced (much more easier) if "zoom to mouse position" is enabled.

discip, 2024-01-18:
I took it for granted, that everyone uses this as default.
That's why I didn't even think about mentioning it. 😅

liftbag, 2024-01-18:
> I think this can be reproduced (much more easier) if "zoom to mouse position" is enabled.

Thank you for the tip. I will adopt this mode while the problem is solved, even if zooming on the cursor is definitely more convenient and faster.

liftbag, 2024-03-30:
> Just tested the 2.0 beta, and the bug still persists. The Zoom to mouse position setting cannot yet be used.

2.0.0 stable too.

liftbag, 2024-04-02:
The pull request #4844 seems seems to work well.

closed 2024-07-10 by github-actions[bot]