OrcaSlicer/OrcaSlicer #3619: Import Model/STL Keyboard Shortcut Does Not Work

doubletaco, 2024-01-11:
### OrcaSlicer Version

2.7.0

### OS version

Windows 11

### Printer

Creality K1 Max

### How to reproduce

According to the File menu, Import > Import 3MF/STL/STEP etc. option has a keyboard shortcut of `Ctrl I, I`

1. Open OrcaSlicer
2. Navigate to the Prepare tab
3. Press `Ctrl + I` followed by `I`

### Actual results

The Open File dialog box does not appear. No action is taken.

### Expected results

The Open File dialog should open as if the user clicked the Import 3MF/STL/STEP etc. option in the File menu.

### Project file & Debug log uploads

[project and configs.zip](https://github.com/SoftFever/OrcaSlicer/files/13906956/project.and.configs.zip)

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13906981/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-01-11:
I don't see this two key press process you're describing: Controli, then i again.  Where are you getting that from?

`Cntrl+i` opens the import file dialog.
Also, `Cntrl+Shift+i` works if you're trying to use a capital I as shown in the menu

[image]

doubletaco, 2024-01-11:
Ah adding Shift did it.

The confusion came from how I interact with the program. I'm blind, so I
use a screen reader. Unless I read characters one-by-one it doesn't
differentiate between capital and lower case letters in the speech output.
Perhaps a clearer shortcut text would be "Ctrl + Shift + I" which is much
more common to encounter.

tlhintoq, 2024-01-11:
Glad its sorted for you.
Kinda odd though as both `cntrl + i` and `cntrl + shift + i` work on my windows 10 setup.
Makes me wonder if its yet another problem with Windows 11 in general.  Or an interaction with your accessibility software & settings.