OrcaSlicer/OrcaSlicer #3627: z offset causes issues and buldges in slicing model

jxer, 2024-01-12:
### OrcaSlicer Version

1.9

### OS version

10

### Printer

A1

### How to reproduce

simply changing z offset in the nozzle type from 0 causes weird slicer images, which actually show up in the print, setting to 0 causes no issue

### Actual results

slicer gets screwy
https://i.imgur.com/phgRrqA.gif

### Expected results

only first layer z should be adjusted

### Project file & Debug log uploads

https://i.imgur.com/phgRrqA.gif

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2024-01-12:
No solution for you but a couple of workarounds:
1. This is related to arc fitting. It does not appear to be compatible with the z offset feature (i.e. if you disable it the artefacts are gone)
2. As you are using a BBL printer I'd recommend you change your z offset in the printer start code. This can be done as per screenshot below. Search for G29.1 in your start code and amend accordingly. 

In the code below I raise the Z by 0.01 if I am printing ABS/ASA, keeping it at 0 if I am using the PEI or Engineering plate and lowering the nozzle by 0.04 if I am using a textured PEI sheet.

[image]

```
{if (filament_type[initial_no_support_extruder]=="ABS" || filament_type[initial_no_support_extruder]=="ASA")}
G29.1 Z{0.01} ; ABS/ASA print - raise first layer slightly
{else}
{if curr_bed_type=="Textured PEI Plate"}
G29.1 Z{-0.04} ; for Textured PEI Plate
{else}
G29.1 Z{-0.00} ; set to 0 offset
{endif}
{endif}
```

jxer, 2024-01-12:
{if (filament_type[initial_no_support_extruder]=="ABS" && filament_type[initial_no_support_extruder]=="ASA")}

this would be "and" correct?

im just looking for the syntax, i want it to be petg and smooth

jxer, 2024-01-12:
this is what i came up with @igiannakas lmk what you think

; Custom Gcode to deal with Smooth PEI and PETG squish
{if (filament_type[initial_no_support_extruder]=="PETG" && curr_bed_type=="Smooth PEI Plate / High Temp Plate")}
G29.1 Z{0.04} ; PETG - raise first layer slightly
{else}
{if curr_bed_type=="Textured PEI Plate"}
G29.1 Z{-0.02} ; for Textured PEI Plate
{else}
G29.1 Z{-0.00} ; set to 0 offset
{endif}
{endif}

igiannakas, 2024-01-12:
> this is what i came up with @igiannakas lmk what you think
>
> ; Custom Gcode to deal with Smooth PEI and PETG squish {if (filament_type[initial_no_support_extruder]=="PETG" && curr_bed_type=="Smooth [...]

Think it looks OK! Try it out in the gcode viewer - you should see the line applied. If you’re happy then send it :)

jxer, 2024-01-12:
> Think it looks OK! Try it out in the gcode viewer - you should see the line applied. If you’re happy then send it :)

doesn't seem to like the plate type, can you test if you have a sec?

igiannakas, 2024-01-12:
I’ll take a look in the evening when back home

jxer, 2024-01-12:
pretty sure it has something to do with the "/" any idea what language this is written in?

jxer, 2024-01-12:
so this works but its hackish, would prefer to know how to deal with the syntax for the smooth plate (this is in orca slicer)

{if (filament_type[initial_no_support_extruder] == "PETG" &&(curr_bed_type != "Textured PEI Plate" && curr_bed_type != "Cool Plate" && curr_bed_type != "Engineering Plate"))}
G29.1 Z{0.04} ; PETG - raise first layer slightly
{else}
{if curr_bed_type=="Textured PEI Plate"}
G29.1 Z{-0.02} ; for Textured PEI Plate
{else}
G29.1 Z{-0.00} ; set to 0 offset
{endif}
{endif}

igiannakas, 2024-01-13:
Try this. Used elsif to condense the if statements and the plate type is High temp plate internally, so use that.

```
; Custom Gcode to deal with Smooth PEI and PETG squish
{if (filament_type[initial_no_support_extruder]=="PETG" && curr_bed_type=="High Temp Plate")}
G29.1 Z{0.04} ; PETG - raise first layer slightly
{elsif curr_bed_type=="Textured PEI Plate"}
G29.1 Z{-0.02} ; for Textured PEI Plate
{else}
G29.1 Z{-0.00} ; set to 0 offset
{endif}
```

igiannakas, 2024-01-13:
If I am to make a recommendation, the below may be a bit more robust, assuming that you want your PETG to have less squish for all plates:

```
; Custom Gcode to deal with Smooth PEI and PETG squish
{if (filament_type[initial_no_support_extruder]=="PETG" && (curr_bed_type=="High Temp Plate" || curr_bed_type=="Engineering Plate"))}

	G29.1 Z{0.04} ; PETG - raise first layer slightly for high temp and engineering plates

{elsif (curr_bed_type=="Textured PEI Plate" && filament_type[initial_no_support_extruder]=="PETG")}

	G29.1 Z{-0.02} ; for Textured PEI Plate and PETG use slightly less squish

{elsif (curr_bed_type=="Textured PEI Plate")}

	G29.1 Z{-0.04} ; for Textured PEI Plate and not PETG use the default squish

{else}

	G29.1 Z{-0.00} ; for non textured plates and not PETG dont use any offset

{endif}
```

igiannakas, 2024-01-13:
I hope you've replaced the existing offset in the gcode, right?

jxer, 2024-01-15:
Thanks for the help on this, I am all set. The plate name caught be off-guard, where is that file that shows the name?

igiannakas, 2024-01-15:
> Thanks for the help on this, I am all set. The plate name caught be off-guard, where is that file that shows the name?

No idea… i just output it in gcode and copied it from there

jxer, 2024-01-15:
do you have a a1? i appear to have lost my wipe on the brush, not sure why though, its now wiping like 2mm above it, and my offset is like .08 guess i need to look into it, i will just print another brush though, this works to well if its screwing with the brush wipe

igiannakas, 2024-01-15:
No I have an X1. You can restore the original gcode and then paste the change in the set offset line. It shouldn’t cause this. 

Happy to take a look at the gcode if you could paste the original and new one here

jxer, 2024-01-15:
its not gcode based, its a physical issue, the offset commands in the gcode are run after the nozzle wipe command, the physical plate is bending down a touch (very new machine, not sure how they don't see this in testing) and causing it to miss, looks to be a standard hard set -5 on the z, orca could prolly deal with this by giving us an adjustable down height during the wipe, I may just make a taller brush for it,

how long does it typically take for him to look at requests, i found another bug in the color changing algorithm setting the cooling fan to 100, bambu studio is not doing this.

I want to step back and say thank you for the direct help on this, its rare to get REAL help from people, generally I have to figure it out, most don't want to dig in and just give boilerplate responses.

closed 2024-04-23 by github-actions[bot]

jxer, 2024-04-23:
Disappointing

siegmour, 2025-03-25:
I have the exact same issue, the model appears deformed in the slicer (version 2.2.0). This appears to only be a visual bug, as the model is printing correctly.

The code @igiannakas posted does not seem to be working for me. I tried settings the offset to 100.00mm just for testing purposes, but it didn't seem to do anything. Using the Z-Offset option in Orca does work (it starts printing 10cm above the plate) but exhibits the reported bug of deforming models.

@jxer Did the G-Code ever work for you?

igiannakas, 2025-03-25:
Not sure what printer you’re using but if you want to adjust z offset in the slicer disable arc fitting as it’s not compatible with the feature.

If you’re using a klipper  printer there is an alternative way to do this vs my code above that works for bambu machines.

siegmour, 2025-03-26:
@jxer Can you please post the code which was working for you? I am indeed using a Bambu printer.

As @igiannakas reported, disabling "arc fitting" under the profile quality settings in Quality -> Precision actually fixes the display of the model with Z offset. However it still feels like a bug, because I did a test print with arc fitting turned on (and the bugged out display) and it printed as expected. It seems only the display is affected, and not the actual print.

jxer, 2025-03-26:
@siegmour 

this is in starting gcode, remove the similar lines.

; Custom Gcode to deal with Smooth PEI and PETG squish
{if (filament_type[initial_no_support_extruder] == "PETG" &&(curr_bed_type != "Textured PEI Plate" && curr_bed_type != "Cool Plate" && curr_bed_type != "Engineering Plate"))}
G29.1 Z{0.11} ; PETG - raise first layer slightly
{else}
{if curr_bed_type=="Textured PEI Plate"}
G29.1 Z{-0.00} ; for Textured PEI Plate
{else}
G29.1 Z{-0.00} ; set to 0 offset
{endif}
{endif}

siegmour, 2025-05-02:
@jxer Thank you for the reply! Something isn't working on my side, and I'm not sure why.

I am trying a much simpler code - for example (the +100.00 is just for testing purposes, to verify that it's actually taking into account the values):

`;===== for Textured PEI Plate , lower the nozzle as the nozzle was touching topmost of the texture when homing ==
;curr_bed_type={curr_bed_type}
{if curr_bed_type=="Textured PEI Plate"}
G29.1 Z{-0.02} ; for Textured PEI Plate
{endif}
; for Smooth plate
{if curr_bed_type=="Smooth PEI Plate / High Temp Plate"}
G29.1 Z{+100.00} ; for Textured PEI Plate
{endif}
`

Even if I try just the textured plate part for testing, it's not doing anything (of course, I select the textured plate from Bambu Studio):

``;===== for Textured PEI Plate , lower the nozzle as the nozzle was touching topmost of the texture when homing ==
;curr_bed_type={curr_bed_type}
{if curr_bed_type=="Textured PEI Plate"}
G29.1 Z{-0.02} ; for Textured PEI Plate
{endif}`

This feels very weird, as the simple change above should definitely be working. Setting it through the settings in Orca does work however. Makes me wonder if the original G-Code for textured plate in Bambu Studio even works for me. Have you tested with really large values (like the ones I posted) to make sure that the values are working? Any ideas why it might not be working on my side?

Thank you!