OrcaSlicer/OrcaSlicer #3640: Lost Timelapse function

WimVDK, 2024-01-12:
### OrcaSlicer Version

1.9.0

### OS version

Win 11 Pro 23H2

### Additional system information

13900k, 64gb corsair dominator titanium

### Printer

Bambulab X1C

### How to reproduce

1. Slice file
2. Send to printer
3. Select file on printer

### Actual results

No timelapse option present. 

Video: https://drive.google.com/file/d/1s5VzhPTuIctgMZl1lX1S-NR-nNrenmOh/view?usp=sharing

From what i can find this is because the printer thinks the gcode file is not sent from bambu slicer. Which seems accurate with my project file. Loaded the same 3mf file generated entirely from orcaslicer and load it into Bambu Studio, Slice it and send it to the printer (i never use the print option) the timelapse option IS available.

### Expected results

Timelapse option being available on the LCD after sending

### Project file & Debug log uploads

[dragon.zip](https://github.com/SoftFever/OrcaSlicer/files/13923605/dragon.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

WimVDK, 2024-02-01:
any idea @lostfever

WimVDK, 2024-02-18:
> any idea @SoftFever ?

WimVDK, 2024-03-10:
Additionally i just noticed that thumbnails don't seem to work on the bambu handy app either. But they do work on the printer itself

SoftFever, 2024-03-12:
For my testing, it works well.
Have you tried the 2.0.0 beta?

[image]

WimVDK, 2024-03-12:
What's your settings for the gcode flavor/thumbnail. I'll check.

SoftFever, 2024-03-12:
use the built-in setting

WimVDK, 2024-03-12:
I am using the default 1.9.1 settings (Marlin legacy, default thumbnails settings) haven't touched anything. Just wanna verify my settings match yours.

WimVDK, 2024-03-12:
Here's my settings in OrcaSlicer for the X1C (in regards to thumbnails). With

[image]

The original subject of this post though is the "lost timelapse function", which boils down to any file sent to the printer sd card (via send/send all) does not show the ability to toggle on the X1C LCD to turn timelapse on or off. Seems i deleted that video on my google drive but i can make a new one.

Can't use 2.0.0 beta as that seems to break auto calibration but that's a different issue
[image]

WimVDK, 2024-03-31:
@SoftFever 

Issue persists in 2.0.0 stable. No timelapse (enable/disable) option on X1C LED after sending file to printer

closed 2024-07-07 by github-actions[bot]

Thirtybird, 2024-07-20:
This issue is still present in Orca Slicer 2.1.1