OrcaSlicer/OrcaSlicer #3642: add object labeling to enable skip object feature for failing support material

timboll, 2024-01-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...
any printer that supports "Label objects" for object skipping

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I'd like to be able to cancel printing of support material, using the same methods as skipping any other failing object. sometimes i print with support material that is elective, and improves quality of the print, but could survive without as an alternative to losing 10s of hours of print progress on ~50h+ prints. The use case arises when thin organic tree supports go spaghetti, and risk dragging to other parts of the print.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I've considered adding a proxy object (i.e. `cube.stl`) a find-and-replace edit to gcode to place the object label before `; FEATURE: Support`, but the Bambu Lab P1S and X1C printers that I'm working with don't support skip object without going through the cloud at present. I've considered a python post-processing script and may find time to attempt that, but I'm also uncertain whether the Bambu machines would accept that and behave accordingly.

**Additional context**
Add any other context or screenshots about the feature request here.

ToXIc69, 2024-01-16:
cant find the option in my 1.9 install 
+1 for feature

jurassicplayer, 2024-01-22:
I personally haven't tried it yet, but isn't the feature already in OrcaSlicer?
`Prepare` => `Process` => `Others` => `G-code Output`
https://i.imgur.com/pYBfbRh.png

closed 2024-07-30 by github-actions[bot]

dapetcu21, 2025-01-17:
+1 on this? new versions of klipper support object exclusion, which is useful for partially failed prints:
https://docs.mainsail.xyz/overview/features/exclude-objects

namalahideaway, 2025-12-16:
> 
> For more details about how to exclude objects now that it's set up, see https://docs.mainsail.xyz/overview/features/exclude-objects#enable-exclude_object-module-in-klipper

NO!!

That is object exclusion that stop a part... it is not support specific. did you read his feature request???