OrcaSlicer/OrcaSlicer #3643: Scaling a mirrored object removes the mirroring

ThorsReign, 2024-01-12:
### OrcaSlicer Version

1.9

### OS version

Windows 10

### Additional system information

CPU: 7th gen Intel 7700HQ
GPU: GTX 1060
RAM: 32GB

### Printer

Bambu P1S w/AMS

### How to reproduce

1: Import model
2: Paint model
3: Copy model
4: Mirror the copy
5: Scale the original using the scaling button
6: Scale the copy using the scaling button

### Actual results

The copy will lose the mirroring after being scaled, and will return to original orientation.
[image]
[image]
[image]
[image]
[image]

### Expected results

The mirrored orientation should be preserved, resizing the model should not affect the orientation or mirroring of the model.
[Bug.zip](https://github.com/SoftFever/OrcaSlicer/files/13924682/Bug.zip)

### Project file & Debug log uploads

[Bug.zip](https://github.com/SoftFever/OrcaSlicer/files/13924685/Bug.zip)
[tutk_log_iotc_30196.txt](https://github.com/SoftFever/OrcaSlicer/files/13924688/tutk_log_iotc_30196.txt)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2024-01-17:
Yeah that's weird... I could reproduce, will take a look.

closed 2024-01-21 by SoftFever

ThorsReign, 2024-03-06:
@Noisyfox @SoftFever This issue was not resolved by this update - I was able to reproduce using the above steps on Orca 1.9,1.

Noisyfox, 2024-03-07:
> @Noisyfox @SoftFever This issue was not resolved by this update - I was able to reproduce using the above steps on Orca 1.9,1.

It'll be included in 2.0.

PR #3719 (2024-01-21): Retain mirror state when scaling using text input (#3643)
Fix #3643