OrcaSlicer/OrcaSlicer #3644: Ender 3 V3 SE Profile broke

KevLoco, 2024-01-13:
### OrcaSlicer Version

1.9.0

### OS version

MacOS 14.1.1

### Printer

Ender 3 V3 SE w/Sonic Pad running Klipper

### How to reproduce

Just Print a Flow Pass 1 or something else, had this problem and all of the prints.

### Actual results

After a few layers, you notice that the nozzle moves to low an drags the material, the result is a massive rattling noise and a extremely rough surface, I had to cancel most of the prints sliced with this profile, I don't know if it is over extrusion or the Z-Axis doesn't move as expected. Tried a few days to fix this, didn't found a solution with this profile.

### Expected results

none of the above. Ender 3 V3 KE Profile works great with adjusted Speed, so at the moment you can use this if you have the same problem.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bistory, 2024-01-13:
I see you are using a Sonic Pad, so your problem comes probably from a missing calibration.
VE profile is made for a stock VE, which works great.
Try first to calibrate your extruder ;-)

bingo-i, 2024-01-22:
I'm having the same problem.
I've tried a ton of different settings, but I can't get the
but the z-axis is still not working correctly and the
and scratches the bed.
Prior to 1.9.0 I did not have an SE profile, so I was printing optimally with good calculations.
Now that I have the SE profile and use it, I am having problems.

Ender 3 V3 SE - using Nebula Pad (Klipper)

bistory, 2024-01-22:
> I'm having the same problem. I've tried a ton of different settings, but I can't get the but the z-axis is still not working correctly and the and scratches the bed. Prior to 1.9.0 I did not have an [...]

Like said before, you guys are not using a stock printer, so don't expect it to work as is. You need to adapt your gcode like you probably did previously.

KevLoco, 2024-01-22:
Hi Thomas,

Thank you for your answers, I haven't got it to work.
Before using the SE Profile I guess I had modified the Ender 3 V2 Profile which worked really good and also the Ender 3 V2 is not a Klipper machine as I know.
I could not get the SE profile to work as I tried nearly everything, having always the same problem.

But as mentioned in my first Post, I am using the KE Profile now with a few modifications (acceleration and speeds...) and it works flawlessly.
Couldn't found out what the Problem with the SE profile is but im glad it works with the KE Profile.

I guess it works with stock Ender 3 V3 SE so maybe it is my bad and because it's running Klipper, I don't know.

Bests, Kevin

bingo-i, 2024-01-22:
> Hi Thomas,
>
> Thank you for your answers, I haven't got it to work. Before using the SE Profile I guess I had modified the Ender 3 V2 Profile which worked really good and also the Ender 3 V2 is not a [...]

 What a coincidence!  I also did well with the ender v2 profile before I had a se profile!
 Based on your feedback, we will try again with the KE profile.
 thank you!

KevLoco, 2024-01-25:
Fixed by using the Ender 3 V3 KE profile with the speeds from the SE profile

closed 2024-01-25 by KevLoco