OrcaSlicer/OrcaSlicer #3645: Adjustable speed for 3d mouse such as 3Dconnexion SpaceMouse

tlhintoq, 2024-01-13:
A lot of people in the 3d/CAD/3dprinting space have 3d input devices like the Spacemouse.
Orca even recognizes it and moves in the right directions in response to it, which is more than a lot of the slicers do-so thank you for that.

But the read rate is very very slow with no way to speed it up.  Even when the speed is max'd out through the device's control panel the response to it in Orca is snail speed.
[image]

Its a bit faster in rotation than in fore/back or up/down.  Ranking response time from best to worst:
1. Rotation (almost usable speed)
2. Fore/back
3. Left/right pan
4. Z up/down (total crawl speed to the point of unusable)
Adding a setting for `3d mouse read speed` or something like that... `sensitivity`... `speed multiplier`... something... would be of tremendous help.

If you really want to go big while improving 3d mouse support, maybe a few more options like how Lychee lets you invert some behaviors so you can match your CAD modeler would be lovely.
[image]

foreachthing, 2024-01-14:
Press [CTRL]+[M] in OrcaSlicer:
[image]
And you can set your speeds here.

tlhintoq, 2024-01-14:
That's terrific.  **Thank you!**
But if someone didn't know the secret keystroke of `CNTRL + M` how would they know this exists?
I couldn't find it in a menu, in the preferences, right-click {...}  Is it that I'm just that old and blind, or is this totally hidden?

Ohhhh.... Knowing it was a keyboard shortcut made me go look at the `Help | Keyboard shortcuts`.  There it is.
One of those _If you don't know, you don't know to look_ kinda things.
I feel like I just learned the secret handshake and gained access to a new level of the game. (In a good way)

Off to screen grab and make a cheat sheet infographic while I learn these.

Again  **THANK YOU**

closed 2024-01-14 by tlhintoq

jtenniswood, 2024-03-11:
This command doesn't work on the Mac version, as command+m is already used for minimizing windows.
Can I suggest changing the short cut on the Mac?

jtenniswood, 2024-03-13:
> Can you open this dialog?
> image.png (view on web)
> image.png (view on web)

I can open the dialog but I can’t use that short cut because it’s already
assigned to an OS level shortcut so it can’t be accessed

Coffe83, 2024-03-30:
Could we please have this issue fixed. On MacOS you cannot open the settings for 3Dconnection devices with Command+M, because this is the general command for minimizing windows in MacOS. And there seems to be no other way to get to these settings other than the short key. Could you choose another short key, please?

gbomacfly, 2024-05-19:
Same problem here...

SuperTango, 2024-06-05:
This should be addressed in pull request #5598