OrcaSlicer/OrcaSlicer #3650: Infill line multiplier + (dis)connect layer lines [Already in Cura since 2018]

hutje, 2024-01-13:
**Infill line multiplier**

As described in this issue on BambuStudio Github: [https://github.com/bambulab/BambuStudio/issues/1528#issuecomment-1640671376](url)

[image]
[image]

It would be really useful to be able to have the option to have a multiplier value for the infill lines. Sometimes you don't want to increase the infill percentage but still want a stronger infill.

**Connect infill lines**
[image]
Another addition to this would be to be able to have a box to enable and disable "connect layer lines". This way you can expose infill and keep it nice looking. As gyroid connects the layerlines at walls. But without walls, it doesn't look good.
As described here:
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/zig_zaggify_infill.md#connect-infill-lines
****

Both these features are already in Cura. It would be nice to have this in OrcaSlicer as well.

igiannakas, 2024-01-13:
You can increase line width - for a 0.4 nozzle you can go easily to 0.7-0.8 which will give you better/ stronfer results than simply extruding two lines. 

CNC kitchen has done a video on it.

igiannakas, 2024-01-13:
https://youtu.be/9YaJ0wSKKHA

hutje, 2024-01-13:
> You can increase line width - for a 0.4 nozzle you can go easily to 0.7-0.8 which will give you better/ stronfer results than simply extruding two lines.
> 
> CNC kitchen has done a video on it.

True, but not all filament would be suitable to do this. As there is more backpressure when extruding. When using soft TPU I would prefer multiple lines.

hutje, 2024-04-13:
Still hope this option will be implemented... It is not an entirely new feature...

Sureiya, 2024-07-16:
I also would really like this feature.

hutje, 2024-07-16:
I don't understand why this feature is not been implemented already. This feature is already available in cura. It is really useful for printing big objects or using flexible filaments.

RF47, 2024-11-11:
This feature is useful, please add.

closed 2025-02-17 by github-actions[bot]

ShaneDelmore, 2025-03-05:
> True, but not all filament would be suitable to do this. As there is more backpressure when extruding. When using soft TPU I would prefer multiple lines.

I don't think backpressure is an issue if you have correctly set the maximum flow rate for the filament, it will then slow down as needed to the correct speed to ensure the flow rate is not exceeded.

hutje, 2025-03-05:
> I don't think backpressure is an issue if you have correctly set the maximum flow rate for the filament, it will then slow down as needed to the correct speed to ensure the flow rate is not exceeded.

I am not asking for reasons why you don't think it is needed.
The fact you don't need the feature, doesn't mean other people do find it useful to be able to use multiple infill lines.

ShaneDelmore, 2025-03-06:
I didn't intend it as a reason why the feature is not needed.  You keep on trying to get people to do free work for you with that winning personality.