OrcaSlicer/OrcaSlicer #3652: Can't add Elegoo Neptune 4 Max

tlhintoq, 2024-01-13:
### OrcaSlicer Version

2.0.0 Dev

### OS version

windows 10

### Printer

Elegoo Neptune 4 max

### How to reproduce

Add printer.
Select Neptune4max as shown
Look in your printer list - Its not there.

[image]

### Actual results

I can add a default Neptune 3 max (as shown in pic)
I can add and customize a Neptune 3 max (N3Max in `---User Presets---`)
But adding an 4 max just doesn't show up in the printer list.

### Expected results

Should be able to add and select a 4 max.

### Project file & Debug log uploads

No project required

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2024-01-13:
Hmmm... I wonder if this is similar to the STILL UNRESOLVED
https://github.com/SoftFever/OrcaSlicer/issues/2467#issuecomment-1890548903

Where there's a problem in the JSON higher up the inheritance chain

bistory, 2024-01-14:
Hi, I tried to reproduce your issue but I wasn't able to...

[image]

tlhintoq, 2024-01-15:
Well.... Maybe I'm doing something differently?

Printers before adding:
[image]

Click the sprocket icon next to printer list to get to this dialog
[image]
Tick the Max4 box then `Confirm`.  Dialog closes.

Then here's the new printer list. {second verse same as the first}
[image]

Is there a config / file / folder I can zip up and include that might shed some light?

tlhintoq, 2024-01-17:
There had a been an ongoing problem with the FSON definition for a parent printer this inherited from.
That was fixed in another bug, so this was fixed for free by the upstream bug fix.

closed 2024-01-17 by tlhintoq

JPinage, 2025-01-19:
For weeks now I can’t get Orca to even work with my N4 Max. Multiple uninstalls and complete work overs. Files won’t transfer over wifi, files off a USB stick reboot my printer. Orca works fine with my N4Pro but won’t with my Max. I can even do work in fluids on my max in the Device tab or Orca and it says all my connections work but I get 502 bad gateway and print host errors, usb uploads reboot my printer, not these options are flawless on my pro.