OrcaSlicer/OrcaSlicer #3654: SET_VELOCITY_LIMIT not reset after first layer

borisb74, 2024-01-13:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

Neptune 3 Pro (Klipperized)

### How to reproduce

1. Printer uses Klipper firmware; "Emit limits to gcode" is disabled
2. Set first layer velocity to a different value than velocity for normal printing (e.g. 500 vs 3500)
3. Generate gcode for a model

### Actual results

For the first layer is included in the gcode as SET_VELOCITY_LIMIT ACCEL=500 ACCEL_TO_DECEL=250, but the value is never set back to 3500.

### Expected results

Expected is that either SET_VELOCITY_LIMIT is not included in the gcode or that it is set back to the original value after the first layer, i.e., SET_VELOCITY_LIMIT ACCEL=3500 ACCEL_TO_DECEL=1750

### Project file & Debug log uploads

[IKEA 365 Lid Quad.zip](https://github.com/SoftFever/OrcaSlicer/files/13928768/IKEA.365.Lid.Quad.zip)
[debug_Sat_Jan_13_17_16_30_2760.log - Kopie.0.zip](https://github.com/SoftFever/OrcaSlicer/files/13928777/debug_Sat_Jan_13_17_16_30_2760.log.-.Kopie.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

borisb74, 2024-01-14:
After changing `initial_layer_acceleration` to 3500 as well I expected the following line in the gcode:
`SET_VELOCITY_LIMIT ACCEL=3500 ACCEL_TO_DECEL=1750`
However, the output remains
`SET_VELOCITY_LIMIT ACCEL=500 ACCEL_TO_DECEL=250`

Unexpectedly neither custom g-code nor any of configuration parameters includes the number 500 for axis acceleration.

bastus1986, 2024-01-16:
I can confirm this is the case.  Why? It seems pretty unintuitive to allow the extruders maximum accel to change my x and y accel rates.

borisb74, 2024-04-16:
Is there any news on this topic?

krause3284, 2024-04-20:
Yeah I just encountered this issue as well. Seems odd that X and Y acceleration are tied to E acceleration. Does adjusting the E acceleration cause any issues?

borisb74, 2024-06-24:
Any updates on this topic?

tayanov, 2024-07-17:
+1 any idea? its trouble only with neptune 3 pro klipper and orca

DarkCeltic, 2024-09-28:
I have run into this issue as well which lead me to here. If I deselect "Enable accel_to_decel" then Accel is changed throughout gcode.

alber70g, 2024-10-07:
Maybe this is interesting for anyone else

I found that the acceleration is set to the printers acceleration `Maximum acceleration for extruding`. I suppose I'll change that one to 5000 to get to the `Normal printing` acceleration speeds.

[image]

closed 2025-01-13 by github-actions[bot]