OrcaSlicer/OrcaSlicer #3656: No z hop when travelling to prime tower - nozzle scrapes on printed parts

Snelso91, 2024-01-13:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10

### Additional system information

CPU: AMD Ryzen 7 3700X
RAM: 32GB
GPU: Nvidia RTX 3060

### Printer

Bambu Lab A1

### How to reproduce

1. Slice the 1st plate of the example 3mf (this bug isn't specific to this one 3mf project though).
2. Go to layer 26 as an example

### Actual results

Near the end of the layer, when it is finishing the gap fill for the object at the bottom of the plate (gcode line 145386 onwards), you can see that the toolhead height is Z5.200 as expected.
From line 145407 to 145423, you can see the inserted conditional gcode for if timelapse is enabled. If the timelapse is enabled, then the printer would correctly execute a spiral z hop before moving to the timelapse position, taking a picture, and then moving to the wipe tower.
However, if you do not enable the timelapse with the checkbox at the start of the print (as I am doing in this case), then as far as I can tell that whole section of gcode between `M622 J1` and `M621` is skipped due to the `M1002 judge_flag timelapse_record_flag` line. Therefore, contrary to what the slicer shows in the preview window, the toolhead does not spiral lift (or perform any kind of z hop), and moves directly point to point from the bottom object to the starting position on the prime tower.
[image]

This means that the nozzle hits multiple printed objects on the way including the fragile support layers on the left and top of the plate at high speed (500mm/s for travel), and it does this for many layers over and over causing an awful scraping sound.
Luckily it has not caused an outright print failure yet, but I am very worried that eventually the nozzle will hit something that has curled up too much and rip part of the print off the bed, or damage the toolhead.

Also, even if I entirely remove the timelapse gcode from the printer profile in the slicer, it still does not z hop, and this time the slicer correctly shows the straight line path that the toolhead actually takes, and that it remains at Z5.200 the whole time between the bottom object and the prime tower.
[image]

### Expected results

The nozzle should z hop when travelling to the prime tower, as it does when leaving the prime tower or during the rest of the print.

### Project file & Debug log uploads

[AMS Lite Top Mount.zip](https://github.com/SoftFever/OrcaSlicer/files/13929169/AMS.Lite.Top.Mount.zip)
[debug_Sat_Jan_13_12_40_37_30980.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13929170/debug_Sat_Jan_13_12_40_37_30980.log.zip)
[AMS Lite Top Mount Gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/13929171/AMS.Lite.Top.Mount.Gcode.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

amzaldua, 2024-01-14:
I detect this issue too when the nozzle travels in the same layer between two points in the same line type scratching the print and causing the nozzle to be forced.
This issue is with Orca Slicer 1.9.0 (Mac OS 14.2.1 23C71 Apple M1) and BambuStudio v1.18.2.56 too.

closed 2024-06-20 by github-actions[bot]

mnlkrp, 2024-12-14:
Was this ever patchd? I am experiencing the same issue. 
IMO if there was an option to force the prime tower without activating the timelapse, this would fix the issue.

MrDankle, 2024-12-18:
How is this closed without fix? This issue still exists as clearly outlined by the OP I cannot print anything with tree support using my a1s due to this issue!