OrcaSlicer/OrcaSlicer #3657: When dragging program window across multiple screens. Orca moves very slowly and has to catch up to the movement made.

madassassin98, 2024-01-13:
### OrcaSlicer Version

1.9.0

### OS version

Windows 10.0.19045

### Additional system information

Intel(R) Core(TM) i9-9900K CPU
Z390 AORUS PRO WIFI-CF
64gb Trident Z Royal Ram
EVGA 3090 FTW3
Samsung 970 Evo Pro 500gb + 1tb

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

1. Click and hold window
2. Move window from one screen to another
3. Visualize the lag between the application trying to catch up to where its being moved to.
4. It doesnt matter if a model is loaded or not.

### Actual results

Either the application will crash or it will catch up.

### Expected results

Being able to move from one window to the next.

### Project file & Debug log uploads

[OrcaSlicerLogFiles.zip](https://github.com/SoftFever/OrcaSlicer/files/13929255/OrcaSlicerLogFiles.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

stevvie, 2024-04-07:
This is not limited to multi monitors, I have had this since switching to Orca slicer. I'm a Radeom 5700XT if that helps narrow this down.

foreachthing, 2024-04-08:
This is also not limited to window size. Min or max window size makes no difference.
It's like moving around in honey. Had no crash yet.

Nvidia 461.55, Quadro P2000
Intel(R) Xeon(R) W-2125 CPU @ 4.00GHz
RAM: 32 GB
Windows 10 Enterprise (19045.2251)

stevvie, 2024-04-08:
Seems that SuperSlicer suffers the same thing thats also based on prusa slicer.

closed 2024-07-15 by github-actions[bot]