OrcaSlicer/OrcaSlicer #3658: Set support type per modifier object (allow modifier objects to specify different support)

motley74, 2024-01-13:
**Is your feature request related to a problem? Please describe.**
I would like to be able to add a part modifier and set the support type for that object that is different from main object.

**Which printers will be beneficial to this feature**
Bambu X1C/P1S/P1P, any?

**Describe the solution you'd like**
Would like to be able to set support parameters per object instead of being limited to only global setting. For example, I would like to be able to set support type to Normal by default but then add a part modifier and specify that tree supports be used in that area.

This would allow normal supports be used for areas that can be supported better using normal supports, like large flat areas not far from the build plate; but then specify tree supports be used for areas that would normally require supports to start on the main object instead of the build plate, like tops of openings or overhangs inside an object.

This would allow for cleaner parts, less post processing, and likely less filament usage for areas where supports normally start on a part instead of build plate which are harder to remove cleanly.

A good example is the AMS Riser created by Ben Levi on Makerworld (https://makerworld.com/en/models/19535#profileId-19420). Parts 3 & 4 have openings that would be best supported by tree supports so the supports are not started on the part being printed but also have a large flat area that is only a few mm above the build plate and, imo, are better supported by normal supports as tree supports are problematic when they are that short and require such a large surface area to be supported.

jlacroix42, 2024-02-11:
That feature is already present in PrusaSlicer IIRC, I was a somehow surprised you can't do this in OrcaSlicer

closed 2024-05-19 by github-actions[bot]

lockheed, 2025-01-09:
Bump again. That's a pretty critical feature.

LancerGM, 2025-02-28:
Bump. I am relatively new to Orca slicer, and was pretty satisfied with all the features I discovered. This made me all the more surprised this feature is missing. Given that it would be a very intuitive feature.
For some models, painting all supports manually would be too much work, but auto-generating supports will generate an entire tower just for a tiny feature that did not even need supporting anyway. Being able to exclude just that area would be incredibly convenient.

alexfornuto, 2025-04-01:
I wouldn't normally add a "me too" comment, except that this issue was closed by robots before being looked at by human maintainers, so being as loud as possible about it seems like the thing to do.

handmade0octopus, 2025-04-05:
Worth revisiting. Important feature.

TexturedGuppy, 2025-04-08:
Ended up here looking for this feature, so leaving a comment saying it would be a great feature to have.

kk0425, 2025-05-01:
Also would like this feature added. It really would simplify a lot of prints.

yoru-si, 2025-05-22:
Same here, would love this feature in OrcaSlicer, can save some filament and time in the long run.

VanUsual, 2025-06-04:
I second that!

Voxar, 2025-07-04:
I needed this feature today. Need to stack the plate with variations of hilts and one of the models need the snug grid to not risk falling off the plate but the other would do a lot better with trees.

[image]
[image]

Lexxus17, 2025-07-04:
Currently struggling with this at the moment. Being able to use different support types in a model would be game changer level amazing!!!

confuzled-dev, 2025-08-29:
interested in this feature. would solve some workflow problems for me