OrcaSlicer/OrcaSlicer #3665: Extrusion in the air above the prime tower (V1.9.0)

andronick83, 2024-01-14:
### OrcaSlicer Version

V1.9.0 Official Release

### OS version

Windows 10

### Printer

ELEGOO Neptune 4

### How to reproduce

1. Add filament change
2. Add prime tower
3. Slice

### Actual results

On the last layer, extrusions occur above the prime tower.

[image]
[image]

### Expected results

-

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bistory, 2024-01-14:
Hi,
I saw the same problem on latest build, manual filament swap, enabled prime tower.

[image]

Here is a project to test...

[Parties Noires boitier avant.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13932507/Parties.Noires.boitier.avant.3mf.zip)

igiannakas, 2024-01-14:
This is the ramming action

danno131313, 2024-01-15:
I am experiencing this issue as well.

> This is the ramming action

This happens regardless if filament ramming is enabled or disabled in printer settings. When enabled it shows a rather large amount, when disabled it shows a single line, but in both cases the amount it extrudes for the last layer is shown in mid-air.

[image]
[image]

Kosmoook, 2024-01-24:
I am experiencing the same issues both with my custom config and with any config that isnt using a sort of AMS from Bambulab. This is a showcase of the issue on my custom Vcore profile. 
I am attaching both gcode and 3mf file in a zip. 
[issue.zip](https://github.com/SoftFever/OrcaSlicer/files/14042279/issue.zip)
[image]

closed 2024-05-01 by github-actions[bot]