OrcaSlicer/OrcaSlicer #3667: Invisible text

real-artswan, 2024-01-14:
### OrcaSlicer Version

1.9.0

### OS version

Ubuntu 22.04.3 LTS

### Additional system information

AMD Ryzen 5 PRO 4650G

### Printer

Bambu Lab X1

### How to reproduce

1. Change any setting in a profile
2. Create new file

### Actual results

Dialog to transfer changes appears, but changed settings names are not recognizable
[image]

### Expected results

The font must be black (or somehow musch darker then background)

### Project file & Debug log uploads

nothing to upload

### Checklist of files to include

- [ ] Log file
- [ ] Project file

henrivdr01, 2024-01-23:
Same here. On Ubuntu.

jamincollins, 2024-02-15:
Same on Arch Linux with v1.9.0.

real-artswan, 2024-02-23:
The same on the curent nightly 2.0.0-dev

JannikWilhelm, 2024-02-26:
Same on Fedora 39, might be related to #3936 and #3675

real-artswan, 2024-03-12:
very bad after all, half of the cool (overall) UX is broken by this

Highcooley, 2024-06-11:
Does anybody know if this has been fixed in the meantime? I only see closed issues regarding this topic. Or did the UI improvements also fix the issue?

real-artswan, 2024-09-11:
fixed to me in 2.1.1

closed 2024-12-19 by github-actions[bot]