OrcaSlicer/OrcaSlicer #3669: Rename/remove custom filament profile

real-artswan, 2024-01-14:
I've named profile in a dumb way, and want to rename it, I know that I can save it with a new name but then I want to remove the old one

**Describe the solution you'd like**
A way to rename a filament profile, and also a way to remove them would be very fine.

[image]
I want to remove/rename this one

real-artswan, 2024-01-14:
Found how to remove profile, but just renaming would be much better

Eldenroot, 2024-01-20:
I agree, I dont understand why it is not possible...

CryPt00n, 2024-03-19:
I´ve created a Feature Request for this topic.

https://github.com/SoftFever/OrcaSlicer/issues/4570

gshpychka, 2024-06-19:
This is still relevant.

OsmosisD, 2024-06-20:
Also this. I've saved it under a new name but can't remove the old one because apparently the new one inherited properties from the old one... There's no checkbox to save it as an independent profile... Only user/project. Annoying.

northmatt, 2024-07-26:
Bandaid solution for y'all. In `%appdata%\orcaslicer\user\default` (There is a shortcut in in OrcaSlicer, to the right of `File` in the top left go through the drop down menu > `Help` > `Show Configuration Folder`) you can find the machine/filament/process config files. Rename the json file, then in the json file replace the values for `name` and `print_settings_id` with the new file name.

ideaalab, 2024-07-26:
Ive tried that, but it will not push the changes to the server. At least didnt work for me.
Comment if you succeed, and how.

mwLabs-eu, 2024-07-26:
Ran into the same problems like @ideaalab 
Playing around with the json files did not work.

Brumes-Wolf, 2024-09-19:
This issue is infuriating, the user should simply be able to chose the name and also decide if something should inherit or not.

MillyRamsey, 2024-11-26:
This is something that is very frustrating for me, too. I want to have my filament list nice and neat, and as it is (creating new filaments all the time with pre-generated names, then trying to rename them), it's a mess!

shmonoff, 2024-12-11:
A workaround that may resolve the issue:
1. Select the filament you want to rename
2. Create a new filament. Make it a copy of the current. Adjust the name (suffix)
3. Delete the old one

8bitbanger, 2024-12-15:
> A workaround that may resolve the issue:

Try it, report back.

ideaalab, 2024-12-16:
I want to rename a parent filament, with several childs :(

gshpychka, 2025-03-17:
> Orca bot: this issue is stale because it has been open for 90 days with no activity.

don't close

real-artswan, 2025-06-25:
> Is anyone working on this, or is this just a place to put concerns to be ignored?

Another button, one simple modal... I wish I had a time to install all the dev stack and search how to work with qt, etc... For a while I save with new name and remove old if need.

OsmosisD, 2025-06-25:
In theory, it should be as simple as fixing a line that forces dependencies. So like, a few lines, tops (assuming it doesn't break anything else - which is entirely possible, unfortunately).

Considering that it's as easy as simply _renaming the file in AppData_, it _should_ be this simple. One would hope.

That said, I also haven't personally encountered the issue with the inability to rename things since 2.0.0. Anyone else able to confirm the problem still exists in the current release?

closed 2025-10-01 by github-actions[bot]

barrelltitor, 2025-10-02:
no don't closee

gshpychka, 2025-10-06:
> Why was this closed?

The comment above explains why.

gshpychka, 2025-10-06:
> > > Why was this closed?
> > 
> > 
> > The comment above explains why.
> 
> Why so obnoxious? Someone stole your cereal this morning?

I'm sorry if you read it as obnoxious, it wasn't meant that way - I was just pointing them to the answer to their question in a pretty dry manner.

gshpychka, 2025-10-19:
> No, it doesn't really explain anything.
> 
> No actual useful communication was posted that would shed some light on why this rather obvious and simple feature is missing or won't be implemented. I assumed that maybe this issue was simply never viewed by devs, but if there's a technical reason I think it would be worth pointing it out? Or if this could be done by the community and devs don't want to do it in the near future, why not mention that? A comment-less bot closure after the anti-user, anti-contributor stale bot bullshit that plagues this site is wild.

The reason why the issue was closed is indeed explained directly: 

> This issue was closed because it has been inactive for 7 days since being marked as stale.

There's nothing else to it. There isn't enough communituy engagement here to keep this open.

TheEvilSkeleton, 2025-10-21:
Hi @gshpychka, please forgive for my intrusion and unsolicited advice.

As a FOSS developer and maintainer, while I understand the sentiment of keeping the issue tracker clean, I would advise you not to have a stale bot, especially one that automatically closes stale issues.

As a user, before opening an issue about something, we often check whether the issue/suggestion was already raised. If/When we see that it was raised in the past but notice that it was closed, we may be misled into thinking that it was rejected and not closed by some bot due to inactivity; even worse, we may not know whether the maintainers (in this case, you) are open for contributions, so we're either left puzzled by the situation, or assume that it was rejected.

As a contributor, we may be discouraged from contributing to the project, due to the (mis)assumption and uncertainty outlined above.

For a situation like this, I would likely create an issue label with the name "Patch Welcome" or "Pull Request Wanted" for suggestions, and remove the bot. This will address those issues, and also give a clear indicator to potential contributors that the maintainers are looking for someone to implement the suggestion, which would also encourage them to contribute to the project.

real-artswan, 2025-10-22:
> [@real-artswan](https://github.com/real-artswan) can you reopen the issue please?
I'm sorry, but I can't, no permissions

SoftFever, 2025-10-22:
> @SoftFever thanks! Also, what do you think about https://github.com/SoftFever/OrcaSlicer/issues/3669#issuecomment-3427058767?

I will read later.
Hit by flu I can barely think now🤪

luminarmedia-dharmesh, 2026-09-07:
This workaround thing is so lame. Its a simple renaming functionality which should have been there from the first.

Cura has had this feature for ages and its so easy and convenient to do.

The ability to rename a profile should be no brainer and should have been there as a default feature like other features.

People do experiment with different setting and would want to save them as different profiles to later compare the finished prints and compare what is different in the profiles. And then eventually would want to use a proper naming for the final profile and then delete other profiles.

Well.. I guess have to stick to using the workaround as of now even in 2026.

I am hoping this simple feature is included soon in future releases.

Felix14-v2, 2026-10-06:
Now you have a better way: use the `Detach from parent` option when saving your profile, and after that you can remove the parent. The only problem is that currently you have to save+detach it twice to apply the changes due to a bug, but at least there's an easy way to perform that now.

[image]