OrcaSlicer/OrcaSlicer #3672: Klipper reports unknown command at start/end of print

bbinnard, 2024-01-15:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

FLSun SuperRacer

### How to reproduce

Klipper displays "Unknown command Start print" at beginning of print, and "Unknown command End print" at end of print. These GCode commands should not be included when Klipper is specified.

### Actual results

Klipper displays "Unknown command Start print" at beginning of print, and "Unknown command End print" at end of print.

### Expected results

Above messages should not occur.

### Project file & Debug log uploads

Problem happens with all files

### Checklist of files to include

- [ ] Log file
- [ ] Project file

quantumfrost, 2024-01-16:
start_print and end_print are customary Klipper start and end macros. You can change it in the slicer: printer settings -> machine g-code

bbinnard, 2024-01-16:
I specified Klipper as the GCode type months ago. Prior versions of Orca did not display the 2 messages. 

[image]

quantumfrost, 2024-01-19:
You need to check the machine g-code tab in that dialog.

bbinnard, 2024-01-19:
You are correct - in that tab I had Start and End statements that were left over from prior use of PrusaSlicer. I removed them.

THanks.

closed 2024-01-19 by bbinnard