OrcaSlicer/OrcaSlicer #3674: TAB key not working.

GAZ082, 2024-01-15:
### OrcaSlicer Version

1.9

### OS version

Linux

### Printer

Ender 6

### How to reproduce

Try to tab in the fields and you will find it does not work.
[image]

### Actual results

Nothing.

### Expected results

Cycle through the fields.

### Project file & Debug log uploads

x

### Checklist of files to include

- [ ] Log file
- [ ] Project file

GAZ082, 2024-03-16:
This still happens in 2.0 Beta.

EvilGremlin, 2024-06-21:
2.1.0 beta, no proper navigation, immense annoyance and waste of time. Personally. i consider lack of proper tab/shift+tab/ctrl+tab/ctrl+shift+tab navigation and broken field order as critical bug. 
Is this Linux-only issue?

closed 2024-09-28 by github-actions[bot]