OrcaSlicer/OrcaSlicer #3675: Tab titles hidden/same color as background in dark mode, with exceptions.

GAZ082, 2024-01-15:
### OrcaSlicer Version

1.9

### OS version

Linux

### Printer

X

### How to reproduce

Go to any area with section tabs and you will find that one active, the title gets hid.

This does not happen with all.

### Actual results

[image]
This is ok!
[image]
Bad again...
[image]
Notes is ok!
[image]

On the sidebar, all bad except notes:
[image]
[image]

### Expected results

See the tab titles.

### Project file & Debug log uploads

X

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2024-01-15:
Seems to work in 2.0 dev

https://github.com/SoftFever/OrcaSlicer/assets/11351506/6d636ae5-dfcd-4f34-819e-5d5ec00f6995

GAZ082, 2024-01-15:
Thanks @tlhintoq . Is this in Linux, Gnome 45 Wayland?

tlhintoq, 2024-01-15:
I'm confused... Either this bug was renamed or I replied to the wrong bug.
That video was for a bug about tabbing from field to field not working - the video shows that focus does correctly shift from field to field.

So did I post to the wrong bug, or is this the same bug with an updated description+title?

GAZ082, 2024-01-15:
> I'm confused... Either this bug was renamed or I replied to the wrong bug. That video was for a bug about tabbing from field to field not working - the video shows that focus does correctly shift [...]

You replied in the wrong issue :D

tlhintoq, 2024-01-15:
Ok.  I'm a dork

henrivdr01, 2024-01-23:
I experience the same issue.  Ubuntu 22.04, Gnome.
I will get the detail on version etc.

GAZ082, 2024-04-23:
Bad bot!

closed 2024-07-30 by github-actions[bot]