OrcaSlicer/OrcaSlicer #3677: Color Change sections put part cooling at 100% not respecting filament profile

jxer, 2024-01-15:
### OrcaSlicer Version

1.9.0

### OS version

11

### Printer

A1

### How to reproduce

put 2 petg colors make a color change, check fan speed shows at 100 during the color change sections, not respecting filament profiles (20-35) for petg

### Actual results

fan at 100% through color change sections

### Expected results

should respect fan speeds, bambu 1.8 is not doing this.

### Project file & Debug log uploads

[image]
[nozzle-container+disc-a1_fan.zip](https://github.com/SoftFever/OrcaSlicer/files/13934760/nozzle-container%2Bdisc-a1_fan.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

dartrax, 2024-01-16:
Could you check if it‘s related to #2600 ?

jxer, 2024-01-16:
> Could you check if it‘s related to #2600 ?

doesn't appear to be, the project file is between the photo's, its only in the color change sections.

dartrax, 2024-01-17:
You are right.
When you set the min and max fans speed threshold of the changed Generic PETG @BBL A1 filament profile back to the defaults it looks right. Still a bug since you lowered the threshold and the fan speed goes high...

dartrax, 2024-04-18:
Bug still present in v2.0.0

jxer, 2024-04-18:
> Bug is still present in v2.0.0 !

disappointing to hear, he hasn't fixed any of the requests I have put in, IDK does no one do color change on petg?

DOT-Nobody, 2024-05-17:
one layer colored:

[image]
[image]

more than one layer colored:

[image]
[image]

jxer, 2024-08-16:
then fix the damn thing

dartrax, 2024-11-15:
> IDK does no one do color change on petg?

I do but this bug only affects certain circumstances so I can work around it.

jxer, 2024-11-15:
> > IDK does no one do color change on petg?
> 
> I do but this bug only affects certain circumstances so I can work around it.

how do you work around it?

dartrax, 2024-11-15:
> When you set the min and max fans speed threshold of the changed Generic PETG @bbl A1 filament profile back to the defaults it looks right.
 
I have my settings in a User preset so I don't override the saved values. I don't know if that is the important difference.

dartrax, 2025-02-15:
unstale

jxer, 2025-02-15:
They haven’t fixed this or commented on this in over a year, I think we are screwed

dartrax, 2025-02-15:
This is open source, if we don't want to fix it ourselves, we have to wait until someone with enough time and skills comes across this problem. I just want to leave it open so that it can be found better and to emphasise that it is not solved, there are still people looking for a fix ;-)

dartrax, 2025-05-17:
unstale

dartrax, 2025-08-17:
unstale

cylabr, 2026-09-10:
I am hitting the exact same bug using the version 2.4.2.

- Model: a cylinder where the top part is painted using a second filament.

[image]

- The fan speed of the last layer is at 90%, same as the value set in the overhang cooling setting.

[image]

- The fan speed of all other layers is at 30%, as it should be.

[image]

Felix14-v2, 2026-10-06:
> nozzle-container+disc-a1_fan.zip

I can't reproduce the issue in 2.5.0. The fan speed is increased, but not to 100%, and it correlates with the layer print time. You have to tweak your filament settings to configure fan behavior correctly, so not a bug.

[image]

jxer, 2026-10-06:
@Felix14-v2 uh 6 people mentioned it was a bug, and it was a bug 2 years ago, has literally nothing to do with layer time, the fan was going to 100 percent on color change and not respecting the setting, i will recheck it again, in a couple days, maybe its been fixed, but you can see where i tested it against bambu  studio at the time and it did not have this issue

closed 2026-10-07 by ianalexis