OrcaSlicer/OrcaSlicer #3678: Add plate-based plateholders for filenames, etc

arrisar, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
When naming an output file for upload, there are several useful placeholders, such as `{input_filename_base}`. However, this does not allow us to set up more complex multi-plate files with easy naming conventions.

**Which printers will be beneficial to this feature**
All printers that are connected for direct uploads.

**Describe the solution you'd like**
Introduce some new placeholder parameters, such as `{plate_name}` and potentially others for better naming convention support. Ability to slice and upload multiple plates at a time wouldn't hurt either 😅 

**Describe alternatives you've considered**
I've done my best to investigate the Orca and Prusa docs, but I'm also human, so I may have missed an existing solution or workaround.

**Additional context**
Here is an example of multiple plates I've set up for a given project (in this case the DUMMY 13 by Soozafone).

[image]

arrisar, 2024-01-15:
OK further, deeper diving into the other issues in the Github revealed that there is _already_ a variable for this, it simply isn't documented (as far as I can tell). In which case, the premise of this issue would be to document this and any other available placeholder _somewhere_.

The only two locations I'm aware of for this reference are the following:

- https://wiki.bambulab.com/en/software/bambu-studio/placeholder-list
- https://help.prusa3d.com/article/list-of-placeholders_205643

tlhintoq, 2024-01-15:
I just guessed at the variable name based on other names.  
I guess I got lucky-been doing this for a while now.
[image]

closed 2024-04-23 by github-actions[bot]