OrcaSlicer/OrcaSlicer #3682: I don't understand how to use Manual Filament Change

dobainier, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
Yes, kinda : there is no wiki to understand how to use this feature.

**Which printers will be beneficial to this feature**
I use a P1P without an AMS

**Describe the solution you'd like**
A wiki would be great.

**Describe alternatives you've considered**
I tried the feature, and I tried to add M600 in the filament change gcode.
But the printer does not pause.
Do we need to add a M600 command, if yes, how do we add it ?

giveen, 2024-01-15:
From my reading, its possible that its not supported without the AMS, and you may just have to do it manually 

https://forum.bambulab.com/t/filament-change-colour-without-ams/5460/15

Ocraftyone, 2024-01-15:
There is a custom Gcode section under printer settings called tool change Gcode. You will want to put a pause/park command in there. You also will enable the manual filament change option. This will pause the print on each filament change allowing you to swap filament.

dobainier, 2024-01-16:
> There is a custom Gcode section under printer settings called tool change Gcode. You will want to put a pause/park command in there. You also will enable the manual filament change option. This will pause the print on each filament change allowing you to swap filament.

Cool, thank you very much.
Do you have te code for this pause/park command please ?

icebear74, 2024-01-23:
on most printers, M600 will work ..

closed 2024-05-01 by github-actions[bot]