OrcaSlicer/OrcaSlicer #3683: Too many solid infill layers with no way to adjust

Joseph-Goergen, 2024-01-15:
### OrcaSlicer Version

1.9.0

### OS version

Mac os M1 pro(arm)

### Printer

Bambu p1p

### How to reproduce

1. Insert primitive box as a slicing model
2. modify top layer and bottom layer settings to 1
3. Slice and observe the solid infill layers

### Actual results

There are too many solid infill layers when slicing and want control of top/bottem layer thickness

### Expected results

For there to be only 1 layer on top/bottom when I tell it to after infill. Theres a specific model that I was using that only used 96 grams of filament in 1.8.1. After updating to 1.9.0 it jumped to 111 grams. When observing and testing, modifying the top/bottom layer did not change the amount of filament being used and noticed the high amount internal solid infill being used.

### Project file & Debug log uploads

[Cube.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13938415/Cube.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kisslorand, 2024-01-27:
Set "Top shell thickness" and "Bottom shell thickness" to 0.

Joseph-Goergen, 2024-02-03:
@kisslorand Thanks!

closed 2024-02-03 by Joseph-Goergen