OrcaSlicer/OrcaSlicer #3686: Crash when selecting variable layer height

aleung, 2024-01-15:
### OrcaSlicer Version

1.9.0

### OS version

Ubuntu 22.04

### Additional system information

OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) HD Graphics 4600 (HSW GT2)

### Printer

Bambu Lab A1 mini

### How to reproduce

1. Load model (any)
2. Activate Variable Layer Height

### Actual results

Crash immediately, without the dialog popping up.

### Expected results

No crash

### Project file & Debug log uploads

No information printed in the log file at crash.

Console has the output:
```
[appimage-binfmt-bypass/bin] ERROR: child exited with code 11
```

### Checklist of files to include

- [X] Log file
- [ ] Project file

aleung, 2024-01-15:
The same issue had been fixed in PrusaSlicer 2.6.0 : https://github.com/prusa3d/PrusaSlicer/issues/7274

The issue is related to OpenGL with some Intel Mesa drivers. PrusaSlicer fixed it by using OpenGL 3.3.

I confirm that OrcaSlicer doesn't crash when disable hardware-accelerated by starting with environment variable LIBGL_ALWAYS_SOFTWARE=1:

```
LIBGL_ALWAYS_SOFTWARE=1 ./OrcaSlicer_Linux_V1.9.0_017e49b85560b9c7b15ec6b509c9cf66.AppImage
```

Noisyfox, 2024-01-15:
Unfortunately there is no easy way to switch to OpenGL 3.3 core profile, as we currently still uses some legacy OpenGL APIs which is no longer supported in 3.3.

closed 2024-04-23 by github-actions[bot]