OrcaSlicer/OrcaSlicer #3688: Z bands on cylinder models

uki-3D, 2024-01-15:
### OrcaSlicer Version

1.9.0 stable

### OS version

Win10

### Printer

Bambulab A1

### How to reproduce

1. Open Orca
2. Load attached file
3. Slice with default Bambu A1 and 0.2 Standard profiles
4. Change ONE parameter - Wall Generator to ARACHNE

I noticed that with the Wall Generator set to Classic this pattern is different but still occurs.
I noticed this only in the case of PETG-CF, it probably isn't visible on other materials, but that doesn't mean it isn't there.

### Actual results

The print shows horizontal stripes that exactly match the model in the slicer.

### Expected results

The print shows horizontal stripes that exactly match the model in the slicer.

There should be no visible artifacts.

### Project file & Debug log uploads

[image]
[A1_FILAMENT_SPOOL_HOLDER_EXTENDER.zip](https://github.com/SoftFever/OrcaSlicer/files/13941148/A1_FILAMENT_SPOOL_HOLDER_EXTENDER.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

uki-3D, 2024-01-15:
[image]
I forgot to add photo. Sorry.

igiannakas, 2024-01-16:
This is most likely cooling related and or minimum layer time related.

Hammerfest, 2024-01-16:
> This is most likely cooling related and or minimum layer time related.

If the print match's the slice how is this cooling or layer time related?

Honest question, if you look at the slice it shows patterns that came out similar in the print, so the g-code sent that pattern and it printed in that pattern so it just doesn't seem possible, especially with that very repeatable occurrence.

igiannakas, 2024-01-16:
Oh I see what you mean, they are so small I missed them on the screenshot.

try reducing the slicing tolerance, most likely the model has some weirdness going on with its polygons

uki-3D, 2024-01-16:
> try reducing the slicing tolerance, most likely the model has some weirdness going on with its polygons

Did You mean exporting (to STL) tolerance?
If so, it's probably a bull's-eye! If I change REFINEMENT (Fusion 360) to MEDIUM, the slicer works correctly, but the model is slightly angular instead of "round". Maybe something could be done about this problem.
I always export STL files in HIGH REFINEMENT to ensure as much detail as possible.
Thanks for the tips.

igiannakas, 2024-01-16:
You can filter out small issues like this by adjusting the slicing tolerance - under quality -> resolution. Try increasing it or decreasing it a bit and check for the effect.

ps. In general it’s best to export into STL from fusion and allow the slicer to generate the model.

uki-3D, 2024-01-16:
Yes. I've changed it to from 0.012 to 0.001 and now it's good, at least on preview. I will check it later. 
Thank You!

closed 2024-04-23 by github-actions[bot]