OrcaSlicer/OrcaSlicer #3705: Force quit on filament preset title

Svisty12, 2024-01-16:
### OrcaSlicer Version

1.9.0

### OS version

Mac OS 13.2.1

### Additional system information

Attempting to change the title of a filament profile to a new name, force quits the app. 

### Printer

Bambu Labs A1 Mini

### How to reproduce

1. Click to edit the filament preset
2. Click on the save icon next to the current filament title 
3.  Rename the title to desired name 
4. Force quits 

### Actual results

After the above steps, same issue of force quit. 

### Expected results

Should let me edit the preset to the name desired, and keep editing in the slicer software. 

### Project file & Debug log uploads

[tutk_log_iotc_65032.txt](https://github.com/SoftFever/OrcaSlicer/files/13953919/tutk_log_iotc_65032.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

CHunzinger, 2024-01-17:
Yes, I can confirm this Bug on MacOS 14.2.1.
Every saving process of the settings(Filament/PrintSettings/etc.) triggers this error

ichibana, 2024-02-26:
Same here on macOS 14.2.1 and OrcaSlicer 1.9.1

closed 2024-06-03 by github-actions[bot]