OrcaSlicer/OrcaSlicer #3713: Extrusion rate smoothing does not work when moving from perimeter to infill

devosray, 2024-01-17:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

Ender 3 V3 SE

### How to reproduce

1. Create new project, add a calibration cube. [This model as an example.](https://www.thingiverse.com/thing:1278865l)
2. In the "Speed" section, ensure "Extrusion Rate Smoothing" is set to a value greater than 0. In this example, I have it set to 8.
3. Slice and look at the flow color scheme

### Actual results

Observe that when the head moves along the perimeter, the flow rate is slowly adjusted as it is ramping up and down. But when the print head moves from the slower perimeter to the faster infill, there is now gradual adjustment from slow to fast. The flow rate jumps from 3mms^3 to 15mms^3 directly  after the travel.

[image]

### Expected results

I expect the flow rate to be slowly ramped up, even after a short travel move. This is the result in Prusa slicer:
[image]

### Project file & Debug log uploads

[xy
[debug_Wed_Jan_17_10_27_33_12184.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13961136/debug_Wed_Jan_17_10_27_33_12184.log.zip)
zCalibration_cube.zip](https://github.com/SoftFever/OrcaSlicer/files/13961132/xyzCalibration_cube.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2024-01-18:
It's because the nozzle retracts and travels, hence there is no pressure in it to smooth out. The behaviour from Prusa was intentionally changed because of this, as there is no reason to equalise pressure there.

devosray, 2024-01-18:
Thank you for the explanation, glad to hear it is intentional. I'm having issues with under extrusion after the travel move but I'll see if I can find the root cause.

closed 2024-01-18 by devosray

igiannakas, 2024-01-18:
> Thank you for the explanation, glad to hear it is intentional. I'm having issues with under extrusion after the travel move but I'll see if I can find the root cause.

Try three things:

disable reduce infill retraction under “others”

slightly increase your retraction value (slightly- 20% or so).

increase your travel speed and acceleration