OrcaSlicer/OrcaSlicer #3714: 1.9.0 Selected Model not visible - Paint On Seams/Supports

jbeardon, 2024-01-17:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11

### Printer

P1P

### How to reproduce

Select model.
Select any of the 'paint on' modes - seams, supports etc

### Actual results

Selected model disappears.

### Expected results

Selected model should be the only one visible.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13961154/log.zip)
[Cube.zip](https://github.com/SoftFever/OrcaSlicer/files/13961190/Cube.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2024-01-17:
This has been fixed in latest nightly build.

CaptainRoot, 2024-01-21:
#3758 Still present in 1.9.0 Release

KuroObi72, 2024-02-03:
Problem immer noch da!!! Bei Nähte, Stützen und Farben nur grauer Bildschirm

closed 2024-05-11 by github-actions[bot]