OrcaSlicer/OrcaSlicer #3715: Wiki: Calibration page lacks test interpretation

fangly, 2024-01-17:
### OrcaSlicer Version

1.9.0

### OS version

Ubuntu 22.04

### Printer

Bambu Lab P1P

### How to reproduce

1. In Orca Slice, go to calibration, then tutorial. This opens https://github.com/SoftFever/OrcaSlicer/wiki/Calibration in a web browser.
2. Read the page.

### Actual results

For the following calibration tests, it is not specified what to do with the test results, i.e. what to do with the test results, specifically where to enter the optimized values in Orca Slicer:
1.    temp tower
2.    retraction test
3.    orca tolerance test
4.    max volumetric speed

Further, for the tolerance test, there is also no explanation of how to interpret the test and if there is some setting in Orca Slicer that should be modified to improve the tolerance.

Finally, the VFA test is not described at all.

### Expected results

All calibration tests should be described and it should be clear how to modify Orca Slicer parameters based on the test results.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

closed 2024-04-25 by github-actions[bot]

fangly, 2024-05-25:
The issue is still actual.

SphaeroX, 2024-10-07:
Anyone know how to use the VFA?

disconn3ct, 2024-10-09:
> Anyone know how to use the VFA?

https://www.obico.io/blog/speed-test-in-orcaslicer-vfa-a-comprehensive-guide/