OrcaSlicer/OrcaSlicer #3716: Orca seemingly ignores my close_fan_the_first_x_layers filament setting

fangly, 2024-01-17:
### OrcaSlicer Version

1.9.0

### OS version

Ubuntu 22.04

### Printer

Bambu Lab P1P

### How to reproduce

1. Open the attached fan speed scan tower 3MF project with Orca Slicer (version 1.8.1 or 1.9.0). This project is based on the fan speed scan method described at https://www.printables.com/fr/model/366923-scan-tower-calibrate-a-filaments-fan-temperature-s. 
2. Note that the filament settings specify no cooling for the first 1000 layers (close_fan_the_first_x_layers):
[image]
3. Slice the project.
4. In the preview, select fan speed.

### Actual results

Notice that the fan speed varies **within** each floor of the tower. This is not just a preview issue; I verified that the fan speed actually varies within each floor.
[image]

Seemingly, Orca Slicer is not respecting the "No cooling for the first" setting from my filament profile.

### Expected results

Each floor should be printed with a unique fan speed, as is the case with Bambu Studio (version 1.8.2.56 and 1.8.3.89).
[image]

### Project file & Debug log uploads

[2_PLA_fan_scan_100_0%.zip](https://github.com/SoftFever/OrcaSlicer/files/13963741/2_PLA_fan_scan_100_0.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

chippymiller, 2024-01-18:
Same issues here with both 1.9.0 & 1.8.1! Have tried every damn setting and the closest I come to it working is no fan on all layers except for the first layer and I defiantly don't want it on for the first layer either! Same model and same filament on BambuStudio and it will work just fine, but I need/want to use Orca for the extra settings for printing ASA with a lot of overhangs.

KalleAnka021, 2024-03-31:
I have the issue that i run 0 cooling for first layer but then the second layer is 100% when i dont even have 100% in my settings. Cant get rid of that no matter what settings i try some layers will be 100% fan.

I got this problem arount 1.8.1 i think.

fangly, 2024-05-25:
I am used to Orca Slicer being an improvement over Bambu Studio, especially regarding calibration. But this bug is a sticking issue.

fangly, 2024-08-24:
This issue is still a sore to me.

fangly, 2024-10-28:
If this bug were fixed, I could finally uninstall Bambu Studio from my computer.

fangly, 2025-01-27:
I care.

fangly, 2025-04-28:
Hi

johnny-bit, 2025-04-29:
This should be possible to narrow down... since BS works and Orca doesn't work correctly for that, maybe it's possible to find "last working version" and then find where the problem got introduced?

jmmerida, 2025-05-21:
It happens to me too:

[image]

fangly, 2025-06-10:
This issue has been present since at least version 1.6.6 and until the present version (2.3.0).

fangly, 2025-06-10:
Here is an interesting new finding:

1. Open the attached fan speed scan tower 3MF project with Orca Slicer.
2. Delete the object from the build plate.
3. Add a primitive, e.g. a cylinder.
4. Slice the project.
5. Observe that the fan speeds are as expected.
[image]

Conclusions:
- It's not the close_fan_the_first_x_layers filament setting that is not respected.
- Rather, it is the custom layer change G-Code that is not respected or overwritten by some other parameter. Notice how it is particularly prevalent for the layers with high overhang.

fangly, 2025-06-11:
It seems like some spurious M106 commands are issued in the middle of some layers, causing my custom fan speeds to be overwritten:
[image]

coryrc, 2025-06-16:
> Rather, it is the custom layer change G-Code that is not respected or overwritten by some other parameter. Notice how it is particularly prevalent for the layers with high overhang.

You have "Force cooling for overhangs and bridges" checked, so that makes sense... I would also disable "Keep fan always on".

fangly, 2025-08-30:
@coryrc Thanks for the input!

Turning off "Keep fan always on" does not change the issue. However, I can confirm that turning off "Force cooling for overhangs and bridges" produces the desired outcome. :-)

Both of these options are present in Bambu Studio. Thus I would assume the same behaviour in Orca. So, the question becomes, why is this different in Orca? Is is intended? Should it be considered a bug in Orca or in Bambu Studio?

Felix14-v2, 2026-10-06:
Seems to be fixed in the [latest nightly](https://github.com/OrcaSlicer/OrcaSlicer/releases/tag/nightly-builds), could you check it? I can no longer reproduce the problem:

[image]

fangly, 2026-10-07:
Hi @Felix14-v2, I tried the latest nightly (build f9ddb238) and made the same observation as you; the issue seems resolved.

Felix14-v2, 2026-10-07:
Glad to know. You can close the issue now and reopen it at any time if you'll notice this problem again.

closed 2026-10-08 by igiannakas