OrcaSlicer/OrcaSlicer #3720: Brim (Brim-Object gap?) doesn't account for Elephants-foot compensation...this creates a gap between brim and object

PhilBaz, 2024-01-17:
### OrcaSlicer Version

1.9

### OS version

win 10

### Additional system information

good

### Printer

Voron 2.4

### How to reproduce

 Add elephants-foot compensation and brim and slice.... If Brim-Object gap is zero there should be no gap. but there is! This gap is caused be Elephant foot compensation, and can be closed by reducing this to zero. 

[image]

### Actual results

There is a gap

### Expected results

No gap, brim should touch, or be controlled entirely by brim-object gap

### Project file & Debug log uploads

[brim issue.zip](https://github.com/SoftFever/OrcaSlicer/files/13966621/brim.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

PhilBaz, 2024-02-23:
Just bumping this as its a proper bug and should be pretty easy to fix....I think?

PhilBaz, 2024-03-18:
Bumping as this is a pretty major bug.

PhilBaz, 2024-03-22:
Yes, just to clarify here are some images that show the issue. The brim should follow the (elephants foot compensation/ scaling ) base layer.
[image]
[image]

PhilBaz, 2024-04-27:
@cbaugher I have to think this is one of the largest actual bugs in Orca. This is primary functionality breaking.

RoboCrafty, 2024-05-27:
[image]
I still have the gap even with elephant foot compensation set to 0. Brim-Object distance is 0 too. :/

PhilBaz, 2024-05-27:
@RoboCrafty Then you must have a value in the brim offset. That's not a bug. Just the elephant foot compensation.

RoboCrafty, 2024-05-27:
@PhilBaz I don't get it? I thought to remove the gap between the brim and the object the elephant foot compensation had to be set to 0 because of a bug? And here I have the gap even with it set to 0

PhilBaz, 2024-05-27:
@RoboCrafty 
[image]

RoboCrafty, 2024-05-27:
@PhilBaz Yesss I know that 😅, I mentioned I had that set to 0 and there still is a gap.

PhilBaz, 2024-05-27:
@RoboCrafty Ohh, okay sorry. If you have that set to zero as well as elephants-foot then there should not be a gap. Maybe a bug then.

Just making sure you also didn't change any per-object settings?

HeavilyModdedEnder3, 2024-06-03:
V2.1.0 Beta out already, and still not fixed. I have seen some argue that the elephants foot would cause it to bulge into the brim. But it's a simple feature which SuperSlicer has already implemented by allowing a negative brim offset. 

With this negative brim offset and a **large** elephants foot compensation, you can put a modifier down to make your own custom bottom layer patterns, like a carbon fiber pattern infill, and then the brim goes in and counts as the walls to make it look normal.

This way, OrcaSlicer could even have custom bottom surface patterns for infill! The next step after this would be to make something called a "rim" which always puts a wall around the perimeter, even if the rest of the top surface counts as a wall.
[image]
[image]

See how the top of the model counts as a wall? The "rim" would force in another wall to act as a perimeter: making it look cleaner.

Even without the "rim" if you could put in a negative brim offset, without it saying "Out of Range", then the problem would be fixed. In addition, a larger elephants foot compensation could be added to make your own custom bottom surface pattern, with the addition of brims to act as walls. Finally, another feature which could be useful to allow for your own custom top surface patterns would be to make a "rim", which forces another perimeter on, even with walls existing.

PhilBaz, 2024-06-03:
@HeavilyModdedEnder3 Too much and confuses the issue...please don't hijack this bug fix request, start your own thread. This is not what we are asking for. Please delete your post here and start your own thread.

The bug fix is simply that the brim will account for elephants foot compensation so the brim is not detached. That's all.

neo-jayfeather, 2024-07-01:
https://wiki.bambulab.com/en/software/bambu-studio/parameter/elephant-foot
The last two screenshots seem to indicate that BBL does not plan on fixing this issue on their slicer. Just a note, but this does seem fixable by editing the .\src\libslic3r\brim.cpp and maybe the elephants foot compensation one (.\src\libslic3r\ElephantFootCompensation.cpp).

PhilBaz, 2024-07-02:
@neo-jayfeather 

Ahhh, what a horrible decision... It should 'Just work'. Decoupling elephant's foot from brim is obviously the wrong approach.  Very much hope to see this corrected in Orca.

brunetton, 2024-07-14:
Same here
[image]
[image]
(I made elephant foot compensation large to make the problem obvious)

neo-jayfeather, 2024-07-14:
I was reading the [prusaslicer](https://help.prusa3d.com/article/elephant-foot-compensation_114487) wiki and they stated that elephant's foot compensation should squish the filament out enough so that the brim should be properly aligned-- I don't know if that applies, but that may be why no one (these slicers) have changed it.

idmadj, 2024-07-14:
> I was reading the [prusaslicer](https://help.prusa3d.com/article/elephant-foot-compensation_114487) wiki and they stated that elephant's foot compensation should squish the filament out enough so [...]

Sorta makes sense. If the brim doesn't connect, it's theoretically a sign that efc is set too high. 

That being said, elephant foot happens as the weight of the model squishes out the bottom (it is not meant to compensate for poor z-offset). By the time that happens the first layer has cooled down to the temperature of the bed rather than the nozzle's, so not much inter-line adhesion will take place. Also it's entirely possible that lifting will happen before the model gets a chance to elephant foot and connect with the brim. In both cases the brim will not be given a chance to do its job properly.

PhilBaz, 2024-10-15:
Bump...this should be fixed.

PhilBaz, 2024-10-15:
@neo-jayfeather No, this was originaly a Cura feature and there he implementation is correct...IMHO.  The brim should follow the EFC. Sometimes you use EFC to create a psudo chamfer. Also its nice to keep some EFC on your default profile so then using brim you always have to compensate and turn it off.

Also, i don't think its realistic to think of EFC as perfectly compensating so that it will always perfectly meet the brim. And with EFC its always better to use a little more rather than a little less.

In the end Cura is right here. It will be better if its fixed in Orca. Most people probably don't realize that its just not working currently.

CCS86, 2024-10-18:
Beyond that, even with 0 EFC and 0 brim gap, there is still a gap:

[image]

PhilBaz, 2024-10-18:
Just cross posting this here.....

https://github.com/bambulab/BambuStudio/issues/4420

Please guys, this bug is silly.

I see on the bambu side its been labelled and assigned. Maybe bring your issues to the Bambu side, it may get fixed there then make it to Orca down stream.

evilC, 2025-02-11:
PLEASE FIX!!

hotellonely, 2025-04-10:
Bumping.

JPK-MG, 2025-05-28:
Still a bug in version 2.3.0. 
I was hoping for a solution to this, as it renders EFC completely useless when using mouse ears, or brim. In hope of hacking the bug, I attempted to enter a negative number into Brim Gap - doesn't work, though from a coding perspective, it seems like a logical option... allow negative values in Brim Gap and you could counteract EFC's dominion over brims.

serpent213, 2025-08-22:
Fix would be nice indeed!  👍🏻

PhilBaz, 2025-12-04:
@SoftFever 

Sorry to ping you....can you have a look at this one. Thanks for your work!

CCS86, 2025-12-23:
> You all should look at the printed part for a gap, not in the slicer. If there's a gap that means you overcompensated for elephant foot or have a poorly calibrated printing process and/or printer.

It sounds like you don't understand the issue at all. 

It's pretty simple:

brim-object gap: 0
elephant foot: 0
[image]

brim-object gap: 0.15
elephant foot: 0
[image]

brim-object gap: 0.15
elephant foot: 0.2
[image]

All together now:
[image]

This is with the latest nightly and a primitive cylinder. The third version should still have a 0.15 brim gap, but instead it has a 0.35 gap.

CCS86, 2025-12-23:
It's wild that you are still missing the point here. Elephant foot compensation should not create any brim-object gap. But it currently does.

CCS86, 2025-12-23:
> There's only one bug, you are using a feature out if its intended scope. It's clear that arguing here is futile, you want to use a feature out of its scope and get frustrated it doesn't work the way you would want to.

You are making this claim of "using a feature out of it's intended scope". Elaborate on that. I have no idea what you are talking about.

Let's distill this down to something even simpler for you... the brim-object gap is set to 0 here:

[image]

Explain how this is feature is working correctly.

CCS86, 2025-12-23:
Yes, I understand that, and in a perfect world it would work that way. But, 3D printing is far from perfect. Sometimes the amount of EFC needed to correct the bottom of the model ends up leaving you with a brim that isn't attached. At least allowing a negative brim separation value would let people tune to their liking.

PhilBaz, 2025-12-26:
For those of you who wish to see Orcas handling of brim gap brought in line with Cura Prusa etc...

https://github.com/OrcaSlicer/OrcaSlicer/pull/11699#issuecomment-3690462374

closed 2025-12-26 by PhilBaz

kisslorand, 2025-12-26:
So, which ones are the "right" settings?

PhilBaz, 2025-12-26:
@kisslorand 

Well to be relevant to our discussion you need to adjust the brim offset. But the brim is following the compensation, no?

PhilBaz, 2025-12-26:
@kisslorand 

Like we were ever talking about geometry edge cases? That's a terrible case of semantic drift my dude.

kisslorand, 2025-12-26:
I have kids too :)
...and I'm not smart :( 
(I would have kept my mouth shut if I was smart!)

PhilBaz, 2025-12-26:
@kisslorand 

Fair.

Enjoy the new check box!

CCS86, 2025-12-27:
A tangent discussion:

https://github.com/OrcaSlicer/OrcaSlicer/issues/11750