OrcaSlicer/OrcaSlicer #3724: Cooling Fan - Fan speed-up time changes fan gcode in custom 'change filament G-code'

AndiS977, 2024-01-17:
### OrcaSlicer Version

1.9.0

### OS version

OS: Win10 Pro

### Printer

Mambu Lab P1P

### How to reproduce

1. Slice a project with a filament change.
2. Export g-code
3. Change Cooling fan - Fan speed-up time to e.g. 0.2
4. Slice again
5. Export again
6. Compare the fan commands in the filament change section

### Actual results

Fan gcode commands in custom Change filament gcode are altered.
Left: Fan speed-up set to 0, right: fan speed-up set to 0.2
[image]

### Expected results

All fan g-code commands in custom gcodes should remain unaltered as described in the tooltip of the Fan speed-up time input field.

### Project file & Debug log uploads

[Fan speed-up test.zip](https://github.com/SoftFever/OrcaSlicer/files/13969875/Fan.speed-up.test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

53Aries, 2024-01-19:
Can confirm this. I discovered this same issue yesterday.

AndiS977, 2024-03-13:
Bug still present in 2.0.0 beta.

AndiS977, 2024-06-12:
Bug still present in 2.1.0 beta.

closed 2024-09-18 by github-actions[bot]

MrMonkey1302, 2026-03-29:
same issue in v2.4.0 dev