OrcaSlicer/OrcaSlicer #3726: Flow dynamics calibration method "Pattern" crash

BlackTeaLeaf, 2024-01-18:
### OrcaSlicer Version

1.9.0

### OS version

Windows 11 X64

### Printer

Bambu lab X1C & P1S

### How to reproduce

1.click "Calibration"
2.click "Flow dynamics"
3.select printer
4.click "Manual Calibration"
5.select plate type "Textured PEI Plate"
6.select random "Filament For Calibation"
** 7.select method "Pattern"
8.click "Calibrate"

### Actual results

[image]

### Expected results

Start calibration

### Project file & Debug log uploads

[debug_Thu_Jan_18_15_24_30_43844.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13973527/debug_Thu_Jan_18_15_24_30_43844.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

jintaogao, 2024-02-26:
Also happening on 1.9.1

tobilab, 2024-03-23:
Also happening on 2.0.0-beta

Dedoque, 2024-03-24:
Same, -- 2.0.0-dev

PippyT, 2024-05-08:
It's still happening on 2.0.0 stable release. Orca seems to work flawlessly otherwise.

Voyex, 2024-05-14:
This is occurring in v2.0.0 official release for me as well.

truffaut, 2024-06-06:
I can confirm I'm also seeing this in the 2.1.0 beta release on MacOS.

LouisH98, 2024-06-25:
2.1.0 release windows also crashing

dvomang, 2024-06-26:
This has been non-functional for months now...

closed 2024-10-02 by github-actions[bot]