OrcaSlicer/OrcaSlicer #3727: 100% density on the raft top layer

go1den3ye, 2024-01-18:
I'd like to be able to use a raft with 100% on the top layer, like in cura
[image]
[image]

**Which printers will be beneficial to this feature**
FLSUN V400, Creality CR-10s pro

**Describe the solution you'd like**
1 layer of raft does not work with certain larger projects with my printer and PLA+ filament, but when using other slicers like Cura, and having a 100% density, with several layers of raft, everything works fine

**Describe alternatives you've considered**
1 layer raft with first layer density set to 100% doesn't work. alternative has been to use another slicer on certain print projects. However I would prefer to use Orca for all my prints

**Additional context**
I found someone that has previously put in the same request, but it had been closed (case #475 by user Romrider). I have also seen several other people ask about this on reddit and with a few google searches, so I believe this is a feature several users would like implemented

RoguishlyHandsome, 2024-01-28:
when setting up your Raft... in the Advanced portion, set the base pattern spacing to 0. that should give you the same smooth surface that Cura has. I found this setting by trial and error. If you set the same Top surface Interface layers to 0. You will get the same smooth surface. 
[image]
[image]

You must set the Pattern to Rectilinear pattern to get the best results. Give it a go.

closed 2024-02-05 by go1den3ye

RoguishlyHandsome, 2024-02-06:
I think you may be trying to do a 1:1 setup in Orca from Cura. that may not work the way you want it. If you setup 3 layers in Orca then you will have what orca sees as layer 1 b(bottom) layer 2(Middle) and layer 3(top). 
image1: first layer
[image]
Image2: second layer
[image]
Image3: third layer
[image]

RoguishlyHandsome, 2024-02-06:
> I understand. in my example I was trying to show that when I chose middle/top layers in cura, they are all part of the full raft, so 4 layers of 100% density and offset layers before it starts on any interface or support.
> In Orca it only seems to print one layer of 100% density and offset, then it shrinks the next layer to have less offset to the model, making the "top" layers of the raft too small.
> When I print with PLA+ 1 simple layer doesn't seem to adhere to the printer bed that great when printing certain models, but when I try with the cura settings with 4 stacked layers, I have never had it pop off.
>
> Maybe I just need to find another way around it. I still prefer to use Orca as my main slicer. thanks for the help thou!

Orca will create the first layer based on the expansion size your stipulated, then the subsequent layers based on the size of the base that it is supporting. Take a look at this.

 

This picture below is at 5mm expansion, so that only applies to the first layers. Take a look at the two other layers and how close they get to the object. I think it does it to save material. I think.