OrcaSlicer/OrcaSlicer #3729: Flow Calibration: Model changes Scale when selecting different Material

Culain, 2024-01-18:
### OrcaSlicer Version

1.9.0 official

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

Select Calibration -> Flow Rate -> Pass 2
right click on selected parts -> set filament -> filament 4 (in my case)

### Actual results

All the test parts are bigger now.

### Expected results

Parts should not be bigger after changing Filament

### Project file & Debug log uploads

[scaling models.zip](https://github.com/SoftFever/OrcaSlicer/files/13976043/scaling.models.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

somebody-somewhere-over-the-rainbow, 2024-01-20:
same issue occurs when using a larger nozzle. I experimented with 1.0 nozzles recently and the models gets a lot larger overlaying each other when you slice it

closed 2024-04-27 by github-actions[bot]

Daitallica, 2024-04-29:
Getting this issue with 1mm nozzle on ver 2.0.0