OrcaSlicer/OrcaSlicer #3730: Filament edit page - ramming button not visible (dark theme)

Eldenroot, 2024-01-18:
### OrcaSlicer Version

1.9.0

### OS version

Win 11 64bit

### Printer

A1

### How to reproduce

Go to edit filament page and then to multimaterial tab

### Actual results

[image]

### Expected results

Ramming should have different color (text, button).

### Project file & Debug log uploads

---

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Eldenroot, 2024-04-18:
Still valid

yw4z, 2024-06-15:
this one is on my list, i will update with Printer settings > Printable area > Set button

also this one was fully fixed on my design branch but we decided to not fully merge it

closed 2024-09-21 by github-actions[bot]