OrcaSlicer/OrcaSlicer #3731: Add "Infill Line Multiplier" for strengthening infill.

EdwardChamberlain, 2024-01-18:
[image]

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
Multiple infill lines as an option. This option in Cura duplicates infill lines giving them greater stiffness than individual lines. 

**Describe alternatives you've considered**
-

**Additional context**
-

giveen, 2024-01-18:
https://github.com/Ultimaker/CuraEngine/pull/767

kisslorand, 2024-03-28:
This feature would be a nice one to have in Orca Slicer.

mysteryjeans, 2024-05-12:
The only feature I missed from Cura

nilesjohnson, 2024-07-01:
I was looking for this option and found this issue.  It is indeed a useful option.

ZZZXXX0110, 2024-07-26:
Just to add to the comments that this feature is not only useful for strength, it is also extremely important for building fire resistant parts with fire retardant material (such as Poly maker's PolyMax PC-FR), because typically fire retardant rating for any material is only guarenteed above a certain minimal thickness value (such as 1.5/3mm for UL94 HB rating, the lowest UL94 rating for polymer fire retardant capability), which for 3D printed parts would often be the thickness of their infill. Without customer extruders that have ridiculously high nozzle diameter, the only way to print them would be to use multi-line infill, especially when 100% infill is not an option for high temperature materials (like PC) due to significantly more internal stresses during cooling that could lead to warping.

colorado1876, 2024-07-30:
Please add an infill line multiplier.  This is also a great option for cosmetic designs where an infill pattern is desired for appearance, but one can choose the infill size by selecting density, but also maintain structural integrity of the matter by thickening it.  

As OrcaSlicer is based on Prusa Slicer, it is hard to believe this feature has been omitted, particularly as it does not seem to be difficult to implement.

carrierwave42, 2024-08-25:
I want this too...

technophreak, 2024-09-01:
+1 for me too !

kikomoth, 2024-10-13:
please add :)

kikomoth, 2024-10-13:
+1 please add

RF47, 2024-10-25:
+1 please add

nilesjohnson, 2024-10-28:
I found similar issues for Prusa Slicer and Super Slicer:
https://github.com/prusa3d/PrusaSlicer/issues/7044
https://github.com/supermerill/SuperSlicer/issues/4213

On the Prusa Slicer issue, someone has posted a [proof of concept](https://github.com/prusa3d/PrusaSlicer/issues/7044#issuecomment-2027990079) and their [code changes](https://github.com/profOnno/PrusaSlicer/commit/90862360de111571272ebbb7b8e0f89f8f841805).  So far it doesn't appear that these have been merged anywhere.

su-sd, 2024-11-23:
There are also requests for the bambu slicer:
https://forum.bambulab.com/t/missing-infill-tuning-multiply-infill-lines-walls-angles/6930/12

rooiejoris, 2024-11-23:
I typically use the line multiplier with lightning infill. Making the infill lines connected and infill walls to 2 you get a 'free' outer wall and more sturdy lightning structure. Especially useful for (xxxl) large prints.

gtavo, 2024-12-30:
This wold be really nice...

Mrak-IW, 2025-01-06:
This feature will be usable in a quite unexpected way. You can use it to create a grate in some solid pieces without huge efforts in a CAD. You can just create a model for modifier. Or even use some standard slicer primitives. And set for it top/bottom/wall thickness to 0 and increase infill thickness. In this case you'll get a hole in the piece with infill only in it.
[image]

hutje, 2025-02-23:
Why is Cura the only slicer with this option? This is really useful and needed to make stronger and more rigid object. Not all materials do well by increasing the infill percentage.

I also mentioned this here: https://github.com/SoftFever/OrcaSlicer/issues/3650#issue-2080193632
It makes much more stable objects and will help stiffening flexible filaments without increasing the infill percentage.
[image]

malarcky, 2025-03-12:
Please add this feature; it would be very useful when printing objects without top and bottom layers. It would also be highly beneficial for 3D-printed airplane wings, where weight and strength are the most important factors.

DavidGom1, 2025-03-12:
+1 for me too!

RF47, 2025-04-28:
https://github.com/SoftFever/OrcaSlicer/pull/9432

dewi-ny-je, 2025-05-01:
> Why is Cura the only slicer with this option? This is really useful and needed to make stronger and more rigid object. Not all materials do well by increasing the infill percentage.
> 
> I also mentioned this here: [#3650 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/3650#issue-2080193632) It makes much more stable objects and will help stiffening flexible filaments without increasing the infill percentage. [image]

You are increasing the infill percentage, since you multiply each line. You are just not reducing the spacing between lines.

RF47, 2025-05-01:
Update: I managed to achieve the precise offset of the gyroid curves.
[#9432](https://github.com/SoftFever/OrcaSlicer/pull/9432)

nilesjohnson, 2025-05-02:
> Update: I managed to achieve the precise offset of the gyroid curves. 

Amazing!!!  I'm not in a position to test or review, but I can tell you I'm very happy to see progress :)  Any chance your same approach could be used for other infill patterns?

RF47, 2025-05-02:
> > Update: I managed to achieve the precise offset of the gyroid curves.
> 
> Amazing!!! I'm not in a position to test or review, but I can tell you I'm very happy to see progress :) Any chance your same approach could be used for other infill patterns?
I'm going to try to implement it on infill patterns that don't intersect, such as rectilinear, crosshatch, 3D honeycomb, and aligned rectilinear.
For intersecting infills, it's necessary to generate polygons, and I don't yet know how to do that—I'm open to any advice you may have.

RF47, 2025-05-11:
Update: Added support for several linear infill patterns.
https://github.com/SoftFever/OrcaSlicer/pull/9432

[image]

[image]

[image]

[image]

technophreak, 2025-05-11:
You're awesome! 😎

rooiejoris, 2025-06-02:
Really cool...! however... : )
The idea about the double infill is to reduce starts/stops/travels.
In Cura the infill line is 1 continuous line

[image]

But great work! Please remember I am not a programmer, (many times) I have no idea how difficult some things are (or how easy some other things are...)

cheers / joris

rooiejoris, 2025-06-02:
well, for large objects large travel moves are more complicated due to oozing and 'curled up' end which can be hit during travel
this can cause that the print can be hit from the platform (I experience this with the single line lightning infill and not with the double line connected infill...)
For small prints it doesn't matter that much, for large print it does.
Especially when using large(r) nozzle sizes.

The nice side effect is that you also get an extra wall 'for free'.

RF47, 2025-06-02:
> Really cool...! however... : ) The idea about the double infill is to reduce starts/stops/travels. In Cura the infill line is 1 continuous line
> 
> <img alt="Image" width="797" src="https://private-user-images.githubusercontent.com/1224023/450034708-8d03e713-51a1-41d6-a11d-51ef2ee02b5d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDg4NjM3MTgsIm5iZiI6MTc0ODg2MzQxOCwicGF0aCI6Ii8xMjI0MDIzLzQ1MDAzNDcwOC04ZDAzZTcxMy01MWExLTQxZDYtYTExZC01MWVmMmVlMDJiNWQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDYwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA2MDJUMTEyMzM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2U0MzIxNzZhMjM0NDc2NzA0Yzk4N2NjNmE3M2I2NzkzZjMzYjliOWVjMDQyN2E3ZDAzMDFlZDYzZjE5MDk5NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.x04wcdR4xwb_UB51DYhcES0A5haEfzlAlMmXLAXZqx8">
> But great work! Please remember I am not a programmer, (many times) I have no idea how difficult some things are (or how easy some other things are...)
> 
> cheers / joris

I'll try to achieve this (I'm not a programmer either).
You're right, my implementation is sketchy.

Baggles5, 2025-06-22:
I also am looking to have this feature implemented to reduce retractions while printing TPU.

carrierwave42, 2025-07-04:
Woo-hoo! Thank you!

RF47, 2025-07-04:
> Did connect infill lines get implemented?

I couldn't achieve the Cura´s to behaviour for lightning infill, if that's what you mean.

Baggles5, 2025-07-04:
> > Did connect infill lines get implemented?
> 
> I couldn't achieve the Cura´s to behaviour for lightning infill, if that's what you mean.

Yes that is what I'm waiting for.

RF47, 2025-07-05:
Update:
Added support for 2DLattice & 2D Honeycomb
https://github.com/SoftFever/OrcaSlicer/pull/10060

ianalexis, 2025-07-19:
@nubnubbud this is already implemented in orca's nightly build https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds

RF47, 2025-10-18:
Update: beter lightning infill with multiline:
#11053

[image]

closed 2025-12-23 by igiannakas

RF47, 2025-12-28:
> Really cool...! however... : ) The idea about the double infill is to reduce starts/stops/travels. In Cura the infill line is 1 continuous line
>
> <img alt="Image" width="797" [...]

I think I finally did it. #11765

[image]