OrcaSlicer/OrcaSlicer #3735: I can't "delete" any more after tabbing back from the sliced preview without clicking in the window.

ichibana, 2024-01-19:
### OrcaSlicer Version

1.9.0

### OS version

macOS Sonoma 14.2.1

### Printer

Voron V0.2

### How to reproduce

1. Import a file!
2. Slice it.
3. Hit `Tab` to get back to the Prepare view.
4. Hit whatever shortcut you use to "Select All".
5. Try to hit `Delete` to delete the selection.
6. Ya can't do it unless you **click** in the window. Defeats the purpose of keyboard shortcuts IMHO :)

I'm pretty sure this changed in the latest update.

### Actual results

Can't delete without clicking into the window.

### Expected results

Should be able to delete from the keyboard.

### Project file & Debug log uploads

[example.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13983685/example.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2024-01-19:
Just for ref.
2.0 on Window - Not doing that.  So specific to Mac or to 1.9 maybe???

1. Multiple objects.
2. Slice
3. `Tab`
4. `Cntrl+A`
5. `Delete`
6. yep, all gone.

[image]

@ichibana Any chance its the keyboard?  `Delete` being seen as `Backspace`?  Modifier key like `control` maybe stuck down from a sticky keys utility kind of behavior?

ichibana, 2024-01-19:
@tlhintoq I don't think I have any sticky key modifiers turned on. Also, `Delete` still works to remove stuff **after I click into the window again**. I think this wouldn't happen if the key binding itself was wonky.

ichibana, 2024-01-19:
But curious if any other MacOS folks on 1.9 have the issue! It's definitely not preventing use of OrcaSlicer, of course. And overall, I'm quite happy with it :)

closed 2024-04-27 by github-actions[bot]