OrcaSlicer/OrcaSlicer #3738: filament change error

MBogballe, 2024-01-19:
### OrcaSlicer Version

OrcaSlicer 1.9.0-alpha

### OS version

windous 11

### Printer

DIY 

### How to reproduce

I make a color change between layers 5 and 6, but it is changed at the start of the print

### Actual results

it makes the switch in at the start of the print

### Expected results

is a color change between layers 5 and 6

### Project file & Debug log uploads

[image]
[image]
[disk ø250.zip](https://github.com/SoftFever/OrcaSlicer/files/13987482/disk.o250.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

CooperGerman, 2024-01-19:
Hi, T0 at the beginning of your G-Code corresponds to an instruction telling the printer to select the material 0 (in your case the green PETG). 

Whenever you are slicing with a single extruder multi-material environment the initial tool has to be selected at the beginning of the gcode.

Your issue seems not to be a bug and rather an expected behavior. You can verify that you have a T1 instruction later on in your G-Code at layer 5 (I cannot replicate your project since you didn't attach the project file.)

MBogballe, 2024-01-19:
super cool, it makes sense, I was also wondering about the unknown code that came under print (T0)just tried changing the colors of the filament and now it slices properly and I have an m600 when changing to layer 6 (1.2mm)thank you very much for the quick replyhave really become a big fan of orcaslice Fra: CooperGermanSendt: 19. januar 2024 12:36Til: SoftFever/OrcaSlicerCc: MBogballe; AuthorEmne: Re: [SoftFever/OrcaSlicer] filament change error (Issue #3738) Hi, T0 at the beginning of your G-Code corresponds to an instruction telling the printer to select the material 0 (in your case the green PETG).Whenever you are slicing with a single extruder multi-material environment the initial tool has to be selected at the beginning of the gcode.Your issue seems not to be a bug and rather an expected behavior. You can verify that you have a T1 instruction later on in your G-Code at layer 5 (I cannot replicate your project since you didn't attach the project file.)

closed 2024-04-27 by github-actions[bot]