OrcaSlicer/OrcaSlicer #3740: Feature request: Hotend flow benchmark

Argolein, 2024-01-19:
Issue: trying to manually create the gcode file to create CNC Kitchen's Extrusion benchmark results after the first poop into a printer firmware freeze.

I think it would be cool if the benchmark is implemented into OrcaSlicer:
https://github.com/CNCKitchen/ExtrusionSystemBenchmark

Further info: https://www.cnckitchen.com/blog/extrusion-system-benchmark-tool-for-fast-prints

Argolein, 2024-01-19:
> How is this different than the current flow calibration system?

The current calibration relies on the human eye and you can't see a small deviation (e.g. 5% drop in extruded filament).
The downside of CNC Kitchen's benchmark is that you need a precise scale but then you can be very sure that the commanded and actual extruded filament are the same.

DVSVIDEO, 2024-01-20:
Wrong title!
What you are asking for is the hotend **max.** flow.
It is not a calibration, it is a test to determine what is the maximum amount you can extrude from a given material at a given temperature and with a given nozzle.
It pushes out increasingly larger amounts, over a given time, and then weighs them on a scale to get a value where the amount no longer increases as required by the g-code.
You get the maximum flow for which you can set the maximum rate.

closed 2024-04-27 by github-actions[bot]