OrcaSlicer/OrcaSlicer #3744: Z-slider fully engaged in slice Preview.

liftbag, 2024-01-19:
**Is your feature request related to a problem? Please describe.**
I have already opened this issue on [bambustudio github](https://github.com/bambulab/BambuStudio/issues/1347), without success. I also included it in bulk request #3473.
Probably those who have approached 3D printing with BambuStudio will have acquired the automatic mechanisms to avoid the problem described, but I believe that anyone coming from PrusaSlicer or SuperSlicer may find the behavior of the Z-slider rather irritating.

**Describe the solution you'd like**
It would be great to be able to have the Z-slider fully engaged until the mouse button is released.

**Describe alternatives you've considered**
I can't think of any.

**Additional context**
SuperSlicer behavior

https://github.com/SoftFever/OrcaSlicer/assets/14921654/30073ef2-39e2-4952-bc5b-b97f8de2c598

OrcaSlicer and BambuStudio behavior

https://github.com/SoftFever/OrcaSlicer/assets/14921654/2298fbe4-5b35-4eae-8a59-9d851780183f

liftbag, 2024-01-19:
In fact I just found a partial workaround. By moving the detachable sidebar to the right, I get a reasonable margin to avoid bed flips. The problem remains with the extruder slider.

https://github.com/SoftFever/OrcaSlicer/assets/14921654/088492b8-3ff7-474e-a965-11f5b9e09f2e

DVSVIDEO, 2024-01-20:
I must say that this is the fault of your art.
I tried the same thing and it doesn't behave like that. I can't show you a video, but believe me, it doesn't behave like that.
I use a 2 monitor configuration and have tried it on both monitors.
In full screen mode, of course, you can't pull the cursor out of the window.
However, in windowed mode you can move it up and down by dragging it to the left, and it is ineffective when dragged out of the window. The slicing slider then just stops.
Hovering over the window again will cause the slicer to jump to the position where the cursor returned to the window.
It does not start rotating the view.

 Win10 enterprise, i5-4440S, 12Gb RAM, onboard video Intel hd

liftbag, 2024-01-20:
> Win10 enterprise, i5-4440S, 12Gb RAM, onboard video Intel hd

So it seems like an operating system issue? The behavior you describe is fully acceptable and it would be perfect to have it on macOS too.

DARKFiB3R, 2024-01-27:
> I must say that this is the fault of your config. 

How could such a thing be configured?

I think locking the cursor to the scrollbar is the desired behavior, regardless of its various types of unwanted behavour.

DARKFiB3R, 2024-01-27:
This drives me crazy on a regular basis.

It does not help that when the application is in full screen, I cannot access my task bar (which is in "auto-hide" mode), by moving my cursor to the bottom of the screen..

So I have to choose between one pain in the butt, or the other. Fun times :)

DARKFiB3R, 2024-01-27:
>The behavior you describe is fully acceptable

Disagree. It's the lesser of two evils, but it's still useless and annoying.

liftbag, 2024-01-28:
> I disagree. It's the lesser of two evils, but it's still an unexpected behaviour, that is seemingly useless and very annoying.

In fact you are not wrong. If a developer decides to take on this request, it should be to have the z and extruder sliders fully engaged until the button is released even for Windows and Linux users.

liftbag, 2024-05-02:
> Commenting to keep this issue fresh.

Thanks. This behavior is increasingly irritating to me.

dstulken, 2024-05-02:
I can confirm the same behavior as the last video in the original posting.  
I can also confirm that it is a very commonly triggered and frustrating issue in regular usage.
macOS Sonoma 14.4.1, OrcaSlicer V2.1.0-dev (2024-04-28 nightly build).

evilDave, 2024-07-01:
Can confirm this is an issue on macOS (has been for some time) and it is mighty frustrating. I wish I had time to look at it, but I find most of the code very hard to follow.

liftbag, 2024-09-30:
This issue is still present.

DARKFiB3R, 2024-10-26:
> This issue is still present.

And still annoying the shit out of me every day 😥

liftbag, 2024-11-01:
Annoying, just like the overly small handles for moving and resizing the object. The ones in PrusaSlicer are vastly more comfortable to use. I don’t understand the logic behind choosing such tiny handles.

DARKFiB3R, 2024-11-06:
> Annoying, just like the overly small handles for moving and resizing the object. The ones in PrusaSlicer are vastly more comfortable to use. I don’t understand the logic behind choosing such tiny handles.

I think that's worth creating a separate issue about, because it would definitely be another welcome usability improvment.

liftbag, 2024-11-11:
> I think that's worth creating a separate issue about, because it would definitely be another welcome usability improvment.

I had already submitted the [request,](https://github.com/SoftFever/OrcaSlicer/issues/3473) but due to low interest, Orcabot closed it. In fact, it was a three-part request, which I’ve now restructured to give it more visibility. I could do the same with the request for less unpleasant manipulators to use, but it seems, oddly enough, that these topics aren't quite 'significant enough' to the user base."

liftbag, 2024-11-11:
> I think that's worth creating a separate issue about, because it would definitely be another welcome usability improvment.

Oh, and I'm beginning to doubt my ability to adapt to challenges, because after so much time using it, I'm still fighting (and losing) my daily battle to avoid the unwanted effects mentioned here.

liftbag, 2025-02-14:
The issue is still present.

liftbag, 2025-07-10:
This issue is still present and remains quite annoying. Please do not close it due to inactivity

thelegendtubaguy, 2026-10-28:
PR #14999 fixed this.  This should be closed.

evilDave, 2026-10-28:
Can confirm this is fixed (just tested latest nightly build) on macOS.
Excellent work! This was SO frustrating for what is basically a pretty minor issue.
So happy this is fixed.

closed 2026-10-28 by raistlin7447